Ultimaker/Cura #948: left behind gear icons

markwal, 2016-08-25:
This is a pretty minor display glitch, so I could imagine not wanting to fix since the cost/benefit isn't very good, but just in case this kind of polish matters:

Version: I think this is in all 2.x versions, but I'm running master-e6aec0c6 on a Mac at the moment.

Steps:
Run your mouse up and down the advanced side bar where the gear (settings) icon appears (just to the left of the expander icon "<")

Result:
Notice that sometimes the gear doesn't disappear like it is supposed to when the mouse leaves the area.

sedwards2009, 2016-08-25:
yeah, I noticed this one too today. (Windows 10 64bit with git master)

Ghostkeeper, 2016-08-26:
Been there for a while. We've had this reported as internal issue CURA-627, reported in December 2015 :D It's scheduled for fixing in version 3.1 at the moment.

awhiemstra, 2016-09-05:
This is a problem with the hover state tracking of QML. I remember briefly looking at it and not really seeing an obvious reason why it happens, but basically it boils down to the setting button's hover property being true while the button is not actually hovered.

nallath, 2017-07-27:
Can't reproduce this anymore.

closed 2017-07-27 by nallath