Ultimaker/Cura #949: Multiple file importing is disabled/broken on Linux

nickthetait, 2016-08-25:
https://github.com/Ultimaker/Cura/blob/master/resources/qml/Cura.qml#L644
Any details on what this KDE bug is?
Uncommenting the selectMultiple line successfully allows multiple file importing for me on Debian.

nallath, 2016-08-26:
Good question. I honestly can't remember what the bug was at the time. I'll have to ask @awhiemstra to know for sure, but he's on a holiday right now.

thopiekar, 2016-08-29:
I can remember that I reported a issue on this in the past and @awhiemstra also added a issue report to the KDE bug trackers.
The problem here is that people out there might be still using older KDE versions, where this problem appears.

I'll try to find my own issue report here on Github.

thopiekar, 2016-08-29:
Found my first issue report from `3 Aug 2015 ` :scream:
And I thought I began to report the first time around December :sweat_smile: 

Here it is: https://github.com/Ultimaker/Cura/issues/635

nickthetait, 2016-08-31:
Thanks for hunting that down thopiekar. From what I understand of the history:
1) Importing on KDE was broken, so multiple file importing was disabled on Linux
2) KDE has been fixed, but some people could still be using the old/broken one.

Seems like there are more users that could benefit from this feature (all Linux users) than those who would encounter the not-working case (some KDE users). Not sure how a workaround could be made for something like this. :confused:

nallath, 2016-09-01:
I'm greatly in favour of just re-enabling it again. I don't feel that we should remove a feature as important as this because a subset (KDE users with a old version) of a subset (people with linux) can't use it.

thopiekar, 2016-09-01:
Additionally, the maintainers of the distros should care about adding fixes
to their packages (eg. if they are guaranteeing LTS support like *Ubuntu).
I agree with @nallath 😉

Mit freundlichen Grüßen/With kind regards
- Thomas Karl Pietrowski

nallath, 2016-09-01:
I've just tried this for a bit, but there is more stuff that we need to do to fix it. Simply enabling selectMultiple: true again does not allow for multi loading (eg; the file dialog allows you to select multiple files, but if you do & click open, nothing happens.)

nickthetait, 2016-09-01:
It does actually work for me. Running from source on Debian Jessie allows loading several files. What platform did you use nallath? Any idea what is the cause?

nallath, 2016-09-01:
I'm on ubuntu 16.04, but then again, i did install kbuntu once, but switched to gnome for display again. It could be that i'm the one suffering from this issue in the first place.

nickthetait, 2016-09-01:
I just had Ubuntu 16.04 retested and it worked fine.

Ghostkeeper, 2016-09-04:
Perhaps we can get a more granular platform detection and disable multi-selection only for the specific KDE versions? Could such a thing be detected?

thopiekar, 2016-09-05:
Needs a QML expert here. (KDE comes with a set of QML modules.)

awhiemstra, 2016-09-06:
So the original issue is that with the KDE platform dialog, it would not properly clear the list of selected files in between opening the file dialog. This would mean that you load a file, then load another file you get the first file and the second file (so you end up with three files). Disabling it on all platforms is a bit extreme yes, but this was a long time ago. :)

As for detecting platforms, yes in theory it is possible. I would not really want to go that way though as the code for that would be ugly and full of platform-specific magic.

nickthetait, 2016-09-15:
Recently fixed by d0ad654108bfc56a9cb554d95e960e6702ffec67 thanks
:relieved:

closed 2016-09-15 by nickthetait