Ultimaker/Cura #950: The 2.1.3 version tag is weeks off the actual version 2.1.3 from ultimaker.com

CReimer, 2016-08-27:
The 2.1.3 git tag was set on July 27 onto commit 99ec4bff82ed25d2361e9f9deac1c3f1ddfb0e9b
This was strange from the beginning as Cura itself still claims to be 2.1.2 with this commit.

As it turns out version 2.1.3 from ultimaker.com is actually commit 55d2feae57c72bf61a5215a4254ea2ead2e1a186 from August 16.

Could someone please figure out what went wrong here?

Thanks in advance

Ghostkeeper, 2016-08-27:
I put the 2.1.3 tags there when I got the instruction from my boss that we were going to release 2.1.3 that afternoon. While I was doing that, we got a message that, no, actually, marketing wasn't ready to release 2.1.3 yet. That took two weeks. During this time they also wanted to modify the wording of the change log among some other such things.

On the new release day of 2.1.3 I wasn't at work. It seems that my colleagues didn't update the tags at release.

CReimer, 2016-08-27:
Ahh. That explains much.
There doesn't seem to be much difference between these commits. Nevertheless it's confusing.

But I don't think changing the tag is a good idea either as it changes the checksum of the automatically generated tarballs here on github.

For the future it might be better to wait until the version on ultimaker.com is out and add the git tag afterwards.

Thanks a lot.

closed 2016-08-27 by CReimer