Ultimaker/Cura #951: SD not read when 128 files

web-eagle, 2016-08-27:
Hi,
if you have 128 files on the sd-card, it will not display the files on the card

Ghostkeeper, 2016-08-27:
Cura never displays the files on an SD-card. Not unless you navigate a save-to-file or load-from-file dialogue there anyway.

I think you wanted to post this in the repository for the firmware of the Ultimaker 2? https://github.com/Ultimaker/Ultimaker2Marlin/issues

daid, 2016-08-28:
The firmware indeed has a hard limit on the number of files, due to memory constrains.

closed 2016-08-28 by daid

thopiekar, 2016-08-28:
@daid, has this been added to JIRA? In my opinion this should be added to our FAQ or even to our (future) manuals.
I expected the limit around 1000 or higher, so nobody would even hit that ever.

web-eagle, 2016-08-29:
Hi, is it possible to show a message like “limit on files on SD is reached…”?
All the best
Axel

web-eagle, 2016-08-29:
Hi,
you are right – it belongs to the Ultimaker.

Here is another problem:
This part is not sliced.[cid:image001.png@01D201CE.A679DA20]
This part is not sliced.
I attach also the rsdoc.
Best regards
Axel

Ghostkeeper, 2016-09-04:
Hi web-eagle, your image seems to have failed to upload. A part is not being sliced? It's probably best to make a new issue out of that.