Ultimaker/Cura #952: Thin Wall Slicing Issues (Thin Wall Detection)

opensourcefan, 2016-08-28:
I'm using 2.1.3 on Ubuntu 16.04LTS from the PPA to bypass the Python 3.5 issue. 

I'm having an issue with it detecting or slicing thin walled objects. Upon the layer inspection those walls are almost entirely missed. 

[image]

[image]

This is the file, I found it someplace and use it to calibrate my extrusion multiplier by measuring the walls. 
[_40x10.STL.zip](https://github.com/Ultimaker/Cura/files/441254/_40x10.STL.zip)

I also had another file where the author had pre-built supports, those supports are also thin and were missed entirely. 

Is there a setting that I'm missing? My Line Widths are set a 0.4, my nozzle is also 0.4.

opensourcefan, 2016-08-28:
[image]

[image]

markwal, 2016-08-28:
Isn't this more likely to be an issue for the [Ultimaker/CuraEngine](https://github.com/Ultimaker/CuraEngine/issues) tracker?

opensourcefan, 2016-08-28:
I had no idea that they are different things. I have no problem adding it over there if that what is wanted.

nallath, 2016-08-29:
Printing walls that are exactly 0.4 is always a pain in the ass. It's always possible that some weird rounding issues happen, which cause CuraEngine to ignore the lines. 

We might want to change the "ignore everything below 0.4" into ignore everything below nozzle - 0.01 at some point in the future, but for now I'd suggest slightly decreasing the linewidth (0.39 should work)

Ghostkeeper, 2016-09-04:
We have a setting for that, Nallath! :grin: You can set the outer wall inset to -0.01mm. This makes the outer wall inset just a little bit smaller, which fixes rounding issues like this, at the cost of some theoretical precision and strength. And it keeps the actual extrusion the same. That workaround would only move the problem of course: Meshes that are 0.39mm wide would exhibit the same problem. But it would move the problem into an area that occurs much less often, since people like round numbers in their (mechanical) designs.

This is what happens: CuraEngine slices the model to get cross sections of the mesh, which are polygons. It then gives these polygons an inset of ½ * Line Width for the outer wall. Since the nozzle goes centred over the line, this should make the line (with a line width of ½ * Line Width on either side) go right up to the edge of the original polygon.

With a part that's exactly 0.4mm thick and a line width of 0.4mm, it moves both sides of the cross section 0.2mm inwards, thus causing the inner-side and the outer-side walls to overlap exactly. ClipperLib then removes those bits, since they have no surface area, and the piece disappears.
In this case though, the parts are likely not exactly 0.4mm thick because their coordinates are rounded. They are rounded in the file (any file that Cura understands has finite precision) and they are rounded by CuraEngine down to the nearest micrometre. Since the walls are not axis-aligned, the chance is very small that the line is exactly 0.4mm thick. They are probably thinner in this case.

Depending on the joints used by the modelling software, bits at the corners are still left.

closed 2017-11-14 by ChrisTerBeke