Ultimaker/Cura #953: Part display bugged

Scout205, 2016-08-28:
I installed the latest version of Cura on Ubuntu 16.04 LTS.  Everything seems to work (including slicing) but I cannot view the part interface.  There is not display of the printbed or part.  I cant see the print settings and the GUI tool bar on the left (layer view, rotate, etc.) but I just have a bugged image of my desktop displayed instead of a part on a build plate.

fieldOfView, 2016-08-29:
Could you post a screenshot? Reading your report, I am guessing you made a typo and "*can* see the print settings and the GUI tool bar". Since you are using Ubuntu 16.04, am I correctly assuming you are using the PPA-version? Or are you running from source? AFAIK the official deb does not run on Ubuntu 16.04

What GPU do you use, and is the driver up to date?

Scout205, 2016-08-29:
This is what Cura looks for me.
[image]
[image]

fieldOfView, 2016-08-29:
Thanks for the screenshots. How about my the other questions?

Scout205, 2016-08-29:
I download the install for it and used Ubuntu Software Installer to install it.  I also did go and install the PPA dependencies missing.  As far as IK, the driver is update, tho the GPU is an Intel® 945GM x86/MMX/SSE2.

Note:  Cura 15.*  works fine, including the display.  however after reboot it updates itself to Cura 2.1.* (which maybe just me clicking on update not realizing its updating cura)

fieldOfView, 2016-08-29:
@thopiekar, do you have any clue what could be going on here?

thopiekar, 2016-08-29:
Remembers me of the times when I tried to run Cura on armhf. There seems to be something wrong with your GPU driver.

Please run ```glxinfo``` and post the output here.

fieldOfView, 2016-08-29:
Hmm, that Intel 945GM is a 10 year old GPU. It could simply be too old.
http://ark.intel.com/products/27857/Intel-82945GM-Graphics-and-Memory-Controller

thopiekar, 2016-08-29:
That's what I expect in the glxinfo output.. Something like OpenGL 1.2 is only supported, or something like that..

thopiekar, 2016-08-29:
But there is a workaround.. He can run Cura using software rendering.. There is a environment variable which forces libGL to use Mesa to do the rendering. Damn slow, but should work :wink:

thopiekar, 2016-08-29:
I was sure there is a check for the OpenGL version in UM.
-> https://github.com/Ultimaker/Uranium/blob/master/UM/Qt/GL/QtOpenGL.py#L21
In case he is using the same/greater OpenGL version and still gets errors, then we should set an higher version I guess.
https://github.com/Ultimaker/Uranium/blob/2.1/UM/Qt/GL/QtOpenGL.py should make Cura crash as there is ```None``` missing compared to [master]

Scout205, 2016-08-30:
Here is glxinfo

name of display: :0
display: :0  screen: 0
direct rendering: Yes
server glx vendor string: SGI
server glx version string: 1.4
server glx extensions:
    GLX_ARB_create_context, GLX_ARB_create_context_profile, 
    GLX_ARB_fbconfig_float, GLX_ARB_framebuffer_sRGB, GLX_ARB_multisample, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_import_context, GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, 
    GLX_EXT_visual_rating, GLX_INTEL_swap_event, GLX_MESA_copy_sub_buffer, 
    GLX_OML_swap_method, GLX_SGIS_multisample, GLX_SGIX_fbconfig, 
    GLX_SGIX_pbuffer, GLX_SGIX_visual_select_group, GLX_SGI_swap_control
client glx vendor string: Mesa Project and SGI
client glx version string: 1.4
client glx extensions:
    GLX_ARB_create_context, GLX_ARB_create_context_profile, 
    GLX_ARB_create_context_robustness, GLX_ARB_fbconfig_float, 
    GLX_ARB_framebuffer_sRGB, GLX_ARB_get_proc_address, GLX_ARB_multisample, 
    GLX_EXT_buffer_age, GLX_EXT_create_context_es2_profile, 
    GLX_EXT_create_context_es_profile, GLX_EXT_fbconfig_packed_float, 
    GLX_EXT_framebuffer_sRGB, GLX_EXT_import_context, 
    GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, GLX_EXT_visual_rating, 
    GLX_INTEL_swap_event, GLX_MESA_copy_sub_buffer, 
    GLX_MESA_multithread_makecurrent, GLX_MESA_query_renderer, 
    GLX_MESA_swap_control, GLX_OML_swap_method, GLX_OML_sync_control, 
    GLX_SGIS_multisample, GLX_SGIX_fbconfig, GLX_SGIX_pbuffer, 
    GLX_SGIX_visual_select_group, GLX_SGI_make_current_read, 
    GLX_SGI_swap_control, GLX_SGI_video_sync
GLX version: 1.4
GLX extensions:
    GLX_ARB_create_context, GLX_ARB_create_context_profile, 
    GLX_ARB_fbconfig_float, GLX_ARB_framebuffer_sRGB, 
    GLX_ARB_get_proc_address, GLX_ARB_multisample, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_import_context, GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, 
    GLX_EXT_visual_rating, GLX_INTEL_swap_event, GLX_MESA_copy_sub_buffer, 
    GLX_MESA_multithread_makecurrent, GLX_MESA_query_renderer, 
    GLX_MESA_swap_control, GLX_OML_swap_method, GLX_OML_sync_control, 
    GLX_SGIS_multisample, GLX_SGIX_fbconfig, GLX_SGIX_pbuffer, 
    GLX_SGIX_visual_select_group, GLX_SGI_make_current_read, 
    GLX_SGI_swap_control, GLX_SGI_video_sync
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: Intel Open Source Technology Center (0x8086)
    Device: Mesa DRI Intel(R) 945GM x86/MMX/SSE2 (0x27a2)
    Version: 11.2.0
    Accelerated: yes
    Video memory: 192MB
    Unified memory: yes
    Preferred profile: compat (0x2)
    Max core profile version: 0.0
    Max compat profile version: 1.4
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 2.0
OpenGL vendor string: Intel Open Source Technology Center
OpenGL renderer string: Mesa DRI Intel(R) 945GM x86/MMX/SSE2
OpenGL version string: 1.4 Mesa 11.2.0
OpenGL extensions:
    GL_3DFX_texture_compression_FXT1, GL_AMD_shader_trinary_minmax, 
    GL_ANGLE_texture_compression_dxt3, GL_ANGLE_texture_compression_dxt5, 
    GL_APPLE_object_purgeable, GL_APPLE_packed_pixels, 
    GL_APPLE_vertex_array_object, GL_ARB_ES2_compatibility, 
    GL_ARB_clear_buffer_object, GL_ARB_compressed_texture_pixel_storage, 
    GL_ARB_copy_buffer, GL_ARB_debug_output, GL_ARB_depth_texture, 
    GL_ARB_draw_buffers, GL_ARB_draw_elements_base_vertex, 
    GL_ARB_explicit_attrib_location, GL_ARB_explicit_uniform_location, 
[... 88 lines cut ...]
----------------------------------------------------------------------------
0x05d  0 tc  0  16  0 r  y .   5  6  5  0 .  .  0  0  0  0  0  0  0  0 0 None
0x05e  0 tc  0  16  0 r  . .   5  6  5  0 .  .  0  0  0  0  0  0  0  0 0 None
0x05f  0 tc  0  16  0 r  y .   5  6  5  0 .  .  0 16  0  0  0  0  0  0 0 None
0x060  0 tc  0  16  0 r  . .   5  6  5  0 .  .  0 16  0  0  0  0  0  0 0 None
0x061 24 tc  0  32  0 r  y .   8  8  8  8 .  .  0  0  0  0  0  0  0  0 0 None
0x062 24 tc  0  32  0 r  . .   8  8  8  8 .  .  0  0  0  0  0  0  0  0 0 None
0x063 24 tc  0  32  0 r  y .   8  8  8  8 .  .  0 24  8  0  0  0  0  0 0 None
0x064 24 tc  0  32  0 r  . .   8  8  8  8 .  .  0 24  8  0  0  0  0  0 0 None
0x065  0 tc  0  16  0 r  y .   5  6  5  0 .  .  0 16  0  0  0  0  0  0 0 None
0x066  0 tc  0  16  0 r  y .   5  6  5  0 .  .  0 16  0 16 16 16  0  0 0 Slow
0x067 32 tc  0  32  0 r  y .   8  8  8  8 .  .  0 24  8  0  0  0  0  0 0 None
0x068 24 tc  0  32  0 r  y .   8  8  8  8 .  .  0 24  8 16 16 16 16  0 0 Slow
0x069  0 dc  0  16  0 r  y .   5  6  5  0 .  .  0  0  0  0  0  0  0  0 0 None
0x06a  0 dc  0  16  0 r  . .   5  6  5  0 .  .  0  0  0  0  0  0  0  0 0 None
0x06b  0 dc  0  16  0 r  y .   5  6  5  0 .  .  0 16  0  0  0  0  0  0 0 None
0x06c  0 dc  0  16  0 r  . .   5  6  5  0 .  .  0 16  0  0  0  0  0  0 0 None
0x06d 24 dc  0  32  0 r  y .   8  8  8  8 .  .  0  0  0  0  0  0  0  0 0 None
0x06e 24 dc  0  32  0 r  . .   8  8  8  8 .  .  0  0  0  0  0  0  0  0 0 None
0x06f 24 dc  0  32  0 r  y .   8  8  8  8 .  .  0 24  8  0  0  0  0  0 0 None
0x070 24 dc  0  32  0 r  . .   8  8  8  8 .  .  0 24  8  0  0  0  0  0 0 None
0x071  0 dc  0  16  0 r  y .   5  6  5  0 .  .  0 16  0  0  0  0  0  0 0 None
0x072  0 dc  0  16  0 r  y .   5  6  5  0 .  .  0 16  0 16 16 16  0  0 0 Slow
0x073 24 dc  0  32  0 r  y .   8  8  8  8 .  .  0 24  8  0  0  0  0  0 0 None
0x074 24 dc  0  32  0 r  y .   8  8  8  8 .  .  0 24  8 16 16 16 16  0 0 Slow

thopiekar, 2016-08-30:
Looks like there is something wrong with the code I mentioned in Uranium.

Looking at the code I would expect that Cura won't start as we've set OpenGL 2.0 as minimum (`OpenGL version string: 1.4 Mesa 11.2.0`)

Scout205, 2016-08-30:
So I see the issue is my laptop only supports OpenGL 1.4.  The strange thing is, again, only the display for the 3D space is bugged.  I am still able to modify slicing options and save the gcode.  Is there a work around for this or will support be kept at OpenGL 2.*+?

thopiekar, 2016-08-30:
Instead you can use software rendering, yes.

Go to your terminal and type: ```LIBGL_ALWAYS_SOFTWARE=1 cura```

Now the CPU will do all the rendering, thus using Cura will be incredibly slow, but should work at least without any bugs ( fingers crossed ).

Scout205, 2016-08-30:
The terminal command worked.  It is a bit slow, but I can deal.  There is still a bug that you cant dismiss the "Cura Automatically sends slice info.  You can disable this in preferences"; but i believe someone has already reported this.  Thanks for your help.

thopiekar, 2016-08-30:
Great to hear that. I just would prefer to keep this opened as you should get an error message when using OpenGL 1.4 instead of continuing using Cura.

This really shouldn't happen..

awhiemstra, 2016-09-05:
Interesting, it seems Qt allows getting the 2.0 functions object even if the OpenGL context does not support 2.0. Which means the current check is incomplete and we should expand it. :)

awhiemstra, 2016-09-05:
Or maybe this is a problem with PyQt, because from Qt's docs it seems the templated version should never return a valid object if it does not support the proper version.

thopiekar, 2016-09-05:
Nope, the problem in Cura 2.1 is that a parent instance is not given to QMessageBox as a result the function with the check crashes and for a reason let's the user pass. However, using our 2.3 code I tested it and it works while emulating 1.4 hardware with Mesa.
The only problem is that more than 100 QMessageBox windows appear.
Already, added that problem to JIRA, but don't know it's number by mind.

daid, 2016-09-05:
> Interesting, it seems Qt allows getting the 2.0 functions object even if the OpenGL context does not support 2.0. Which means the current check is incomplete and we should expand it. :)

With the LegacyCura, I had the case where the driver reported that the shader functions exist, but any call to them resulted in an "error" return code. Which resulted in me implementing a "try to build shaders else fall back to fixed function pipeline" hack. (Not sure on which hardware/OS this was happening)

Scout205, 2016-09-09:
So I recently acquired a new laptop with Mobile Intel® GM45 Express Chipset.  Now according to terminal, I do have openGL 2.1, HOWEVER I am getting the same error as I did on the other laptop.

thopiekar, 2016-09-09:
Hmm.. Maybe we already have higher OpenGL requirements? 🤔

Ghostkeeper, 2016-09-09:
As far as I recall, one of our Windows 10 test machines has OpenGL 2.0, and it works on there.

We did encounter a memory corruption fault on the graphics memory of an Ubuntu 16.04 machine, that caused layer view to show all sorts of distortions, random lines and triangles. Perhaps that is related?

thopiekar, 2016-09-09:
Seems to be the problem here. Retried with `LIBGL_ALWAYS_SOFTWARE=1 MESA_GL_VERSION_OVERRIDE=2.1 cura` and it works.
@Scout205, Could you also retry using this command?

Scout205, 2016-10-12:
I havent had a chance to try it as that laptop ended up having a meltdown and the OS needs reinstall.  Once I have it back up an running, I will try that.

nallath, 2017-07-27:
I'm closing this due to inactivity. We've not had any other reports of this failing.

closed 2017-07-27 by nallath