Ultimaker/Cura #956: Enhancement: Relocate "Clear Build Platform" Option After Right Click

opensourcefan, 2016-08-29:
I right click to move around a lot and the odd time the dialog will come up and I accidentally hit the clear build platform button. Would it be possible to have the dialog box open a slight distance away from the mouse pointer or something else to prevent an accidental platform clear. 

[image]

nallath, 2016-08-30:
The re-load all objects seems like a better choice. That shouldn't break anything if you accidently hit it.

Ghostkeeper, 2016-09-04:
Made an issue in our internal tracker, so that it can get allocated to a sprint.

Note that it should be possible to undo the clear (Ctrl+Z) as a workaround.

opensourcefan, 2016-09-12:
This can be closed, clear build platform has been swapped out of the way with reload on 2.3.0

Thank you,  not so scary to right click any more.

closed 2016-09-12 by opensourcefan