Ultimaker/Cura #960: Can't change profile on 2.3.0

graham89, 2016-09-01:
It won't let me change profile?

[image]

nallath, 2016-09-02:
Could you provide me with some logs? See https://github.com/Ultimaker/Cura#logging-issues

ChrisRiddell, 2016-09-02:
I hit the same issue, This is the log for me

> 2016-09-02 18:03:22,517 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_logSocketState [165]): Socket state changed to Listening
2016-09-02 18:03:22,517 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (startEngine [55]): Started engine process: C:\Program Files\Cura 2.3\CuraEngine.exe
2016-09-02 18:03:22,517 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_backendLog [88]): [Backend] Calling engine with: ['C:\\Program Files\\Cura 2.3\\CuraEngine.exe', 'connect', '127.0.0.1:49674', '-j', 'C:\\Program Files\\Cura 2.3\\resources\\definitions\\fdmprinter.def.json', '']
2016-09-02 18:03:22,548 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_backendLog [88]): [Backend] 
2016-09-02 18:03:22,548 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_backendLog [88]): [Backend] Cura_SteamEngine version DEV
2016-09-02 18:03:22,548 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_backendLog [88]): [Backend] Copyright (C) 2014 David Braam
2016-09-02 18:03:22,548 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_backendLog [88]): [Backend] 
2016-09-02 18:03:22,548 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_backendLog [88]): [Backend] This program is free software: you can redistribute it and/or modify
2016-09-02 18:03:22,548 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_backendLog [88]): [Backend] it under the terms of the GNU Affero General Public License as published by
2016-09-02 18:03:22,548 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_backendLog [88]): [Backend] the Free Software Foundation, either version 3 of the License, or
2016-09-02 18:03:22,548 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_backendLog [88]): [Backend] (at your option) any later version.
2016-09-02 18:03:22,548 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_backendLog [88]): [Backend] 
2016-09-02 18:03:22,548 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_backendLog [88]): [Backend] This program is distributed in the hope that it will be useful,
2016-09-02 18:03:22,548 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_backendLog [88]): [Backend] but WITHOUT ANY WARRANTY; without even the implied warranty of
2016-09-02 18:03:22,548 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_backendLog [88]): [Backend] MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
2016-09-02 18:03:22,548 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_backendLog [88]): [Backend] GNU Affero General Public License for more details.
2016-09-02 18:03:22,548 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_backendLog [88]): [Backend] 
2016-09-02 18:03:22,548 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_backendLog [88]): [Backend] You should have received a copy of the GNU Affero General Public License
2016-09-02 18:03:22,548 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_backendLog [88]): [Backend] along with this program.  If not, see <http://www.gnu.org/licenses/>.
2016-09-02 18:03:22,564 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Settings\ContainerRegistry.py (load [158]): Could not determine container type for file C:\Users\****/AppData/Local/cura\machine_instances\old\1\Ultimaker+2%2B.cfg, ignoring
2016-09-02 18:03:22,767 - DEBUG - C:\Program Files\Cura 2.3\plugins\MachineSettingsAction\MachineSettingsAction.py (_onContainerAdded [40]): Not attaching MachineSettingsAction to ultimaker2_extended_plus; Machines that use variants are not supported
2016-09-02 18:03:22,923 - DEBUG - C:\Program Files\Cura 2.3\plugins\MachineSettingsAction\MachineSettingsAction.py (_onContainerAdded [40]): Not attaching MachineSettingsAction to ultimaker_original_dual; Multi-extrusion printers are not supported
2016-09-02 18:03:23,220 - DEBUG - C:\Program Files\Cura 2.3\plugins\MachineSettingsAction\MachineSettingsAction.py (_onContainerAdded [40]): Not attaching MachineSettingsAction to ultimaker2_plus; Machines that use variants are not supported
2016-09-02 18:03:36,877 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_logSocketState [165]): Socket state changed to Connected
2016-09-02 18:03:36,877 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_onSocketStateChanged [155]): Backend connected on port 49674
2016-09-02 18:03:36,877 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Controller.py (setActiveView [80]): Setting active view to SolidView
2016-09-02 18:03:37,518 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\cura\Settings\ExtruderManager.py (getMachineExtruders [280]): Tried to get the extruder trains for machine Ultimaker 2+, which doesn't exist.
2016-09-02 18:03:37,518 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\cura\Settings\ExtruderManager.py (getMachineExtruders [280]): Tried to get the extruder trains for machine Ultimaker 2+, which doesn't exist.
[... cut ...]
2016-09-02 18:04:06,791 - CRITICAL - D:\2.3\build\inst\lib\python3.4\site-packages\cura\CrashHandler.py (show [24]): Traceback (most recent call last):
2016-09-02 18:04:06,791 - CRITICAL - D:\2.3\build\inst\lib\python3.4\site-packages\cura\CrashHandler.py (show [24]):   File "D:\2.3\build\inst\lib\python3.4\site-packages\cura\Settings\MachineManager.py", line 619, in setActiveQuality
2016-09-02 18:04:06,791 - CRITICAL - D:\2.3\build\inst\lib\python3.4\site-packages\cura\CrashHandler.py (show [24]): AttributeError: 'NoneType' object has no attribute 'nameChanged'

ChrisRiddell, 2016-09-02:
This issue seems to effect all toggles and drop downs. as nothing happens when trying to uncheck "Print Build Plate Adhesion" and the material dropdown does something funny by selecting two types.

[image]

nallath, 2016-09-02:
I can't reproduce it myself, but i do see some places where we haven't coded defensively enough. I'll add some extra checks that should prevent this behavior.

ChrisRiddell, 2016-09-02:
Okay after talking in IRC I think I found what's happening here.

With cura 2.1.3 and 2.3 installed there is a folder called cura and cura2.3 in the appdata folder, it seems cura 2.3 is looking in the cura folder for the profiles not the cura2.3 folder as uninstalling and deleting the appdata folders for both versions and just installing 2.3 beta there is now just a cura folder in the appdata and its now working fine.

nallath, 2016-09-02:
Seems like something is still going wrong with our version converter. Thanks for looking into it!

shupack, 2016-09-04:
I'm having the same issues, will try Chris's Fix.

fieldOfView, 2016-09-04:
Did you by chance keep a copy of the files so we can study them?

shupack, 2016-09-05:
no, I'm not that forward thinking....

thopiekar, 2016-09-17:
@shupack, @ChrisRiddell Is that problem still existing?

shupack, 2016-09-17:
it's fine, deleting the appdata folder and re-installing worked for me. 

TY.

thopiekar, 2016-09-17:
No problem 😉 Wish you a lot of fun!

closed 2016-09-17 by thopiekar