Ultimaker/Cura #961: What happened to 2.2?

nickthetait, 2016-09-01:
Not a problem, just confused as to why 2.3 was just released when last one was 2.1.x.
And reminder to plz tag this release ;)

nallath, 2016-09-02:
There isn't a really good reason for it. 

We will tag 2.3 once we no longer have a beta release.

nickthetait, 2016-09-02:
haha, thx

closed 2016-09-02 by nickthetait

fieldOfView, 2016-09-02:
Perhaps we'll end up publishing only odd version numbers. Did you ever see a Cura 2.0?

Ghostkeeper, 2016-09-05:
I think what happened was that we started with 2.2 development after 2.1. Then we had a 2.2 beta internally. Then features got added and it was decided to call it 2.3 because it's different than 2.2.

2.0 was called 2.1 because of expectation management; do I remember that right?

I'll try to remember the tagging, and hopefully only tag it after the release is put online :yum: