Ultimaker/Cura #962: Wire Printing Z feedrate and Extrusion Rates wrong

innovo3d, 2016-09-01:
Wire Printing Z feedrate and Extrusion Rates wrong
I tried out wire printing today on my custom built printer.  It looks like it doesn't move in Z, all of it stayed in the same first layer.  It printed the skirt fine, but as soon as it starts the print, it changes the extrusion rate and was barely pushing out filament.  I use 1.75mm diameter filament, without wire printing enabled it prints fine, but it looks like wire printing is not using the correct extrusion rates and Z steps/mm for my printer.

[image]
[image]

nallath, 2016-09-02:
That coudl be the case. We only tested it with 2.85mm. @BagelOrb Could you have a look at this?

BagelOrb, 2016-09-02:
Wire printing was neglected for a year, but when it was on its death bed I rescued it from annihilation.

The extrusion rate problem has been solved.
The difference in extrusion between the skirt and the print is solved.

The visualization has never worked, because wire printing uses diagonal moves which cannot be visualized in our layer view.

I don't know of any problem with Z steps per mm. I did a print after my fixes and it looks just fine.
Can you tell me more about that problem? What did you actually perceive to conclude that the Z steps per mm are wrong?

BagelOrb, 2016-09-02:
PS: I fixed wire printing yesterday, so it will be in the next beta version. It's a shame it's not fixed in the current beta though..

BagelOrb, 2016-09-02:
Note that the Z speed is generally limited by the firmware of your printer. Most printers are not built for rapid movement along the Z axis.

innovo3d, 2016-09-05:
I managed to get it to print.  The only thing that I noticed now is just the feedrates.  It seems like the skirt feedrate is too high and the print feedrate is way too low.  I was using 2.2 though.  I'll try with 2.3 and let you know.

BagelOrb, 2016-09-06:
No need for that. Those wire printing issues have just been solved for the next 2.3 beta release (not in the current 2.3 beta unfortunately)

BagelOrb, 2016-09-06:
The layer view is still a problem however.

ianpaschal, 2018-02-27:
This has not been updated in more than a year so the parts of it which were not fixed are considered won't fix.

closed 2018-02-27 by ianpaschal