Ultimaker/Cura #963: Infill Mesh Layers View

ChrisRiddell, 2016-09-02:
When you check on the Infill Mesh under the Special Modes, the model is grayed out under the layers view and the slider becomes blank.

Entries from the log when Infill Mesh was checked.

> 2016-09-02 21:58:09,417 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]): Exception: An exception occurred while creating layer mesh.
2016-09-02 21:58:09,418 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]): Traceback (most recent call last):
2016-09-02 21:58:09,418 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]):   File "C:\Program Files\Cura 2.3\plugins\LayerView\LayerView.py", line 251, in run
2016-09-02 21:58:09,419 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]):     layer = layer_data.getLayer(layer_number).createMesh()
2016-09-02 21:58:09,420 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]): AttributeError: 'NoneType' object has no attribute 'createMesh'
2016-09-02 21:58:09,421 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]): Exception: An exception occurred while creating layer mesh.
2016-09-02 21:58:09,422 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]): Traceback (most recent call last):
2016-09-02 21:58:09,422 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]):   File "C:\Program Files\Cura 2.3\plugins\LayerView\LayerView.py", line 251, in run
2016-09-02 21:58:09,422 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]):     layer = layer_data.getLayer(layer_number).createMesh()
2016-09-02 21:58:09,422 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]): AttributeError: 'NoneType' object has no attribute 'createMesh'
2016-09-02 21:58:09,450 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]): Exception: An exception occurred while creating layer mesh.
2016-09-02 21:58:09,451 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]): Traceback (most recent call last):
2016-09-02 21:58:09,451 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]):   File "C:\Program Files\Cura 2.3\plugins\LayerView\LayerView.py", line 251, in run
2016-09-02 21:58:09,452 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]):     layer = layer_data.getLayer(layer_number).createMesh()
2016-09-02 21:58:09,452 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]): AttributeError: 'NoneType' object has no attribute 'createMesh'
2016-09-02 21:58:09,483 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]): Exception: An exception occurred while creating layer mesh.
2016-09-02 21:58:09,484 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]): Traceback (most recent call last):
2016-09-02 21:58:09,484 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]):   File "C:\Program Files\Cura 2.3\plugins\LayerView\LayerView.py", line 251, in run
2016-09-02 21:58:09,485 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]):     layer = layer_data.getLayer(layer_number).createMesh()
2016-09-02 21:58:09,485 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]): AttributeError: 'NoneType' object has no attribute 'createMesh'
2016-09-02 21:58:09,487 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]): Exception: An exception occurred while creating layer mesh.
2016-09-02 21:58:09,488 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]): Traceback (most recent call last):
2016-09-02 21:58:09,488 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]):   File "C:\Program Files\Cura 2.3\plugins\LayerView\LayerView.py", line 251, in run
2016-09-02 21:58:09,489 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]):     layer = layer_data.getLayer(layer_number).createMesh()
2016-09-02 21:58:09,489 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]): AttributeError: 'NoneType' object has no attribute 'createMesh'
2016-09-02 21:58:09,517 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]): Exception: An exception occurred while creating layer mesh.
2016-09-02 21:58:09,518 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]): Traceback (most recent call last):
2016-09-02 21:58:09,518 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]):   File "C:\Program Files\Cura 2.3\plugins\LayerView\LayerView.py", line 251, in run
2016-09-02 21:58:09,519 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]):     layer = layer_data.getLayer(layer_number).createMesh()
2016-09-02 21:58:09,519 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]): AttributeError: 'NoneType' object has no attribute 'createMesh'
2016-09-02 21:58:09,550 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]): Exception: An exception occurred while creating layer mesh.
2016-09-02 21:58:09,551 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]): Traceback (most recent call last):
2016-09-02 21:58:09,551 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]):   File "C:\Program Files\Cura 2.3\plugins\LayerView\LayerView.py", line 251, in run
2016-09-02 21:58:09,552 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]):     layer = layer_data.getLayer(layer_number).createMesh()
2016-09-02 21:58:09,552 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]): AttributeError: 'NoneType' object has no attribute 'createMesh'
2016-09-02 21:58:09,584 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]): Exception: An exception occurred while creating layer mesh.
2016-09-02 21:58:09,584 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]): Traceback (most recent call last):
2016-09-02 21:58:09,585 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]):   File "C:\Program Files\Cura 2.3\plugins\LayerView\LayerView.py", line 251, in run
2016-09-02 21:58:09,588 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]):     layer = layer_data.getLayer(layer_number).createMesh()
2016-09-02 21:58:09,589 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]): AttributeError: 'NoneType' object has no attribute 'createMesh'
2016-09-02 21:58:09,616 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]): Exception: An exception occurred while creating layer mesh.
2016-09-02 21:58:09,617 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]): Traceback (most recent call last):
2016-09-02 21:58:09,617 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]):   File "C:\Program Files\Cura 2.3\plugins\LayerView\LayerView.py", line 251, in run
2016-09-02 21:58:09,617 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]):     layer = layer_data.getLayer(layer_number).createMesh()
2016-09-02 21:58:09,618 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]): AttributeError: 'NoneType' object has no attribute 'createMesh'
2016-09-02 21:58:09,632 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]): Exception: An exception occurred while creating layer mesh.
2016-09-02 21:58:09,633 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]): Traceback (most recent call last):
[... cut ...]
2016-09-02 21:58:09,966 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]):   File "C:\Program Files\Cura 2.3\plugins\LayerView\LayerView.py", line 251, in run
2016-09-02 21:58:09,966 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]):     layer = layer_data.getLayer(layer_number).createMesh()
2016-09-02 21:58:09,966 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Logger.py (logException [53]): AttributeError: 'NoneType' object has no attribute 'createMesh'

nallath, 2016-09-02:
The infill mesh should not be visible in the "main" setting screen. It should only be used from the per-object settings screen.

ChrisRiddell, 2016-09-03:
Even doing the select settings on the per-object setting and adding Infill Mesh the layer view stops working.

Ghostkeeper, 2016-09-05:
From the looks of it, you only have an infill mesh and no non-infill mesh. That would create empty g-code, and I think the front-end doesn't handle that very well with its top-layers.

Infill mesh is highly experimental. Nothing has been implemented in the front-end for it yet (except those two settings). But try checking Infill Mesh and setting Infill Density to 80% for one mesh (per model settings), disable mesh collision in the preferences, then load another mesh and make the two meshes intersect.

closed 2017-06-29 by ChrisRiddell