Ultimaker/Cura #964: Slicing appears to restart infinitely

greenlee, 2016-09-02:
When slicing a file that contains and error and if the infill density is set to 100% the slicing progress bar will run to completion then restart. Save to file never enables.
set the infill to <100% and file slices normally.
Same file repaired works correctly.
[Sample files
](https://drive.google.com/folderview?id=0B411HngnbcLWblc0dTNjT2ZpbVk&usp=sharing)

nickthetait, 2016-09-02:
Is this using the new beta build (version 2.3)? And on what operating system does this occur?

greenlee, 2016-09-02:
Sorry meant to include that.
Cura 2.3 beta
MS Windows 10.0.14393 x64

nallath, 2016-09-02:
Even the one with errors slices fine if i set the infill to 100. Could you also post the GUI logs?

greenlee, 2016-09-02:
I would be happy to.  I dont see a log file called GUI where would i find that.
I did post all the log files I found in the user/appdata/local/cura folder [here](https://drive.google.com/folderview?id=0B411HngnbcLWblc0dTNjT2ZpbVk&usp=sharing)

Ghostkeeper, 2016-09-05:
That folder has the log we need. I'd copy the relevant part here, but Google Drive won't let me keep things selected.

The only abnormality I see is that it says that the engine process returned value 1, rather than what I'd expect (0). Would that break things?

awhiemstra, 2016-09-06:
That probably means the engine crashed and is being automatically restarted, which leads to it crashing again since it still tries to slice with the same settings. Not sure why it is crashing though.

greenlee, 2016-09-06:
If there is anything I can provide to help just let me know.

BagelOrb, 2017-07-23:
I take it that this problem doesn't occur any more in the latest release.

closed 2017-07-23 by BagelOrb

greenlee, 2017-07-23:
Correct, I have not run into this again.