Ultimaker/Cura #965: Eject Removable Disk

ChrisRiddell, 2016-09-03:
Application Version: 2.3 beta
Platform: Windows 10

When you click the eject button in the popup from when you saved the file a new popup comes up but it's blanks I'm guessing it should say it's now safe to remove the disk or something?

[image]

shupack, 2016-09-04:
this works on my machine... yes, it says "safe to remove"

Ghostkeeper, 2016-09-05:
Actually, I've seen this often back when we were working on 2.1 when something went wrong with the ejecting. Perhaps you can reproduce it and see if there's anything in the log? https://github.com/Ultimaker/Cura#logging-issues

ChrisRiddell, 2016-09-05:
It only happened once and looking through the log the only thing i see around the time it happened is this but I don't think it is related.

> 2016-09-05 12:54:16,564 - DEBUG - C:\Program Files\Cura 2.3\plugins\CuraEngineBackend\CuraEngineBackend.py (_terminate [186]): Attempting to kill the engine process
2016-09-05 12:54:16,564 - DEBUG - C:\Program Files\Cura 2.3\plugins\CuraEngineBackend\CuraEngineBackend.py (_terminate [186]): Killing engine process
2016-09-05 12:54:16,568 - DEBUG - C:\Program Files\Cura 2.3\plugins\CuraEngineBackend\CuraEngineBackend.py (_terminate [186]): Engine process is killed. Received return code 1
2016-09-05 12:54:16,655 - DEBUG - C:\Program Files\Cura 2.3\plugins\CuraEngineBackend\CuraEngineBackend.py (_onStartSliceCompleted [222]): Sending slice message took 0.09238886833190918 seconds
2016-09-05 12:54:16,655 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_onSocketError [190]): Unhandled socket error Arcus Error (5, native 10054): Could not send message header

Ghostkeeper, 2016-09-05:
That is indeed unrelated. Though if you had trouble with getting stuck in preparing to slice or that it keeps restarting the slicing, that could be what you see here.

thopiekar, 2016-09-17:
@ChrisRiddell Is the problem still existing?

ChrisRiddell, 2016-09-17:
I have not come across it since the frist time.

thopiekar, 2016-09-17:
Ok and thank you for the quick reply :+1:

closed 2016-09-17 by thopiekar