Ultimaker/Cura #969: Profile Settings Don't Match

ChrisRiddell, 2016-09-03:
Application Version: 2.3 beta
Platform: Windows 10

Within the manage profile window on each profile the settings don't match what's on the side settings panel even though there is a note saying "Your current settings match the selected profile."

[image]

fieldOfView, 2016-09-03:
Doh, I thought we had this fixed. Thanks for reporting it is still happening.

shupack, 2016-09-04:
I have this issue as well, also, it's not changing profiles...stuck on low-quality.  I can't select normal or high..

numanair, 2016-09-04:
@shupack Did you install 2.3 while you had 2.1 installed? My imported profiles are uneditable.

shupack, 2016-09-05:
@numanair Yes, I had 2.1 for about 20 minutes (didn't look at the download page closely..)

this fixed my issues:

https://github.com/Ultimaker/Cura/issues/960

ChrisRiddell, 2016-09-06:
@fieldOfView in the current beta what's used when the gcode is built the settings from the side panel or the ones from the profiles window?

nallath, 2016-09-06:
@ChrisRiddell It uses "user_settings" -> Any settings not yet saved into a profile and "quality_changes" -> Any user changes made based off a "quality" and saved into a profile. Those two are merged and saved into a g-code. 
If you don't make any changes, the settings saved should be empty.

fieldOfView, 2016-09-06:
Rely on the settings as shown in the sidebar.

fieldOfView, 2016-09-06:
@ChrisRiddell, could you do us a favor and zip up your profile folder (C:/Users/[your username]/AppData/Local/Cura) so we can have a look?

ChrisRiddell, 2016-09-06:
Okay this is interesting just checked the profiles window and its changed from when I took the screenshot when I opened this report. (but they still don't match)

[image]

@fieldOfView Here is my cura folder from appdata
https://dl.dropboxusercontent.com/u/91917089/cura.zip

fieldOfView, 2016-09-06:
Thanks, that is very helpful, in the sense that it confirms that there is nothing wrong with the files in your archive. The good news is that we seem to have fixed this issue, after the build that went out as the 2.3 beta was made. 

For now, please don't pay attention to what that overview says, we hope to have a new beta build later this week.

closed 2016-09-06 by fieldOfView