Ultimaker/Cura #970: Performance: two transforms when loading a scene

sevaa, 2016-09-03:
When loading a scene, Cura optionally moves it along the Y line so that all Y coordinates a positive, then, if the model is too big or too small, applies a uniform scale.

Both are implemented as general purpose transformations - by left-multiplying vertex coordinates by a matrix. That's a lot of extra unnecessary arithmetic (multuplying by zeros, adding those zeros up). In case of large meshes, this takes a while. The transforms could've been both united together and performed with less operations.

Ghostkeeper, 2016-09-06:
@nallath says in our internal issue tracker:

> That's not true. The scale change sets the SceneNode transformation matrix. Only one operation on all verts is done.

sedwards2009, 2016-09-07:
Unless this is having a noticeable negative impact on the user's experience with Cura, I say close this issue. At the very least can someone use numbers to say what a big model is and what "takes a while" means.

awhiemstra, 2016-09-07:
Both operations actually work on the scene nodes. The only mesh transform we do during loading is moving all vertices so the mesh' bounding-box center is at 0,0,0 . In addition, the two operations are done in two completely unrelated pieces of code: The auto scale is done on model load, the move operation is a continuous process handled by the platform physics class. It would be quite tricky to combine those into a single operation. So I agree with @sedwards2009 and will close this.

closed 2016-09-07 by awhiemstra