Ultimaker/Cura #971: When adding supports, when a skirt is in use, the supports do not begin printing until Layer 2

jterranella, 2016-09-04:
When printing an item that needs supports from bed, if you use a skirt to prime, and said skirt is more than 3mm away from object, supports will not print until Layer 2. Will not adhere to bed at layer 2.

Using a raft for the time being, since the supports will print at first layer of actual model.

Ghostkeeper, 2016-09-05:
Hmm, thanks for the report. We changed some things with this to prevent clipping with brim. Perhaps that caused it.

BagelOrb, 2016-09-05:
You can prevent this from happening by setting the skirt line count to 1 for the time being.
If you need more skirt lines, please increase Skirt/Brim Minimum Length

Ghostkeeper, 2016-09-06:
Fixed in 2.3.

closed 2016-09-06 by Ghostkeeper

BagelOrb, 2016-09-06:
PS: fixed in the next 2.3 beta release - not the current one.