Ultimaker/Cura #972: Getting infill and solid layers in the middle objects.

pulcher, 2016-09-04:
# The problem
I don't know if this is problem with the model or not, but it seems like a bug to me.  

I was trying to print the "Set of 3Penta 2Tri" stl from this posting on Thingiverse: 
https://www.thingiverse.com/thing:1655790

I made a video, but Github says I am gluten, so I put it up on youtube:
https://www.youtube.com/watch?v=_99M6_WlAAQ

# The engine log
Calling engine with: ['/Applications/Cura.app/Contents/MacOS/CuraEngine', 'connect', '127.0.0.1:49674', '-j', '/Applications/Cura.app/Contents/Resources/cura/resources/definitions/fdmprinter.def.json', '']

Cura_SteamEngine version DEV
Copyright (C) 2014 David Braam

This program is free software: you can redistribute it and/or modify
it under the terms of the GNU Affero General Public License as published by
the Free Software Foundation, either version 3 of the License, or
(at your option) any later version.

This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU Affero General Public License for more details.

You should have received a copy of the GNU Affero General Public License
along with this program.  If not, see <http://www.gnu.org/licenses/>.
[ERROR] Unregistered setting mesh_position_x
[ERROR] Unregistered setting mesh_position_y
[ERROR] Unregistered setting mesh_position_z
[ERROR] Unregistered setting center_object
[WARNING]  -s cool_fan_full_at_height="0.2" -s jerk_wall="20" -s experimental="0" -s support_extruder_nr="0" -s cooling="0" -s support_interface_height="1" -s meshfix="0" -s material="0" -s speed_support="60" -s infill_overlap="15" -s wall_line_width="0.45" -s jerk_layer_0="20" -s speed_print="60" -s jerk_print="20" -s blackmagic="0" -s wall_thickness="2.25" -s speed="0" -s dual="0" -s wireframe_flow="100" -s resolution="0" -s speed_wall="30.0" -s machine_settings="0" -s shell="0" -s travel_compensate_overlapping_walls_enabled="True" -s support="0" -s acceleration_support="3000" -s switch_extruder_retraction_speeds="20" -s support_infill_rate="20" -s platform_adhesion="0" -s jerk_support="20" -s retraction_speed="25.0" -s travel="0" -s wireframe_roof_outer_delay="0.2" -s wireframe_roof_drag_along="0.8" -s wireframe_roof_fall_down="2" -s wireframe_strategy="compensate" -s wireframe_drag_along="0.6" -s wireframe_top_jump="0.6" -s wireframe_flat_delay="0.1" -s wireframe_bottom_delay="0" -s wireframe_top_delay="0" -s wireframe_flow_flat="100" -s wireframe_flow_connection="100" -s wireframe_printspeed_flat="5" -s acceleration_wall="3000" -s wireframe_printspeed_down="5" -s wireframe_printspeed_bottom="5" -s wireframe_roof_inset="3" -s wireframe_height="3" -s wireframe_printspeed_up="5" -s magic_fuzzy_skin_thickness="0.3" -s support_conical_min_width="5.0" -s support_conical_enabled="False" -s skin_alternate_rotation="True" -s skin_outline_count="0" -s support_interface_density="100" -s coasting_enable="False" -s draft_shield_height="10" -s draft_shield_height_limitation="full" -s draft_shield_dist="10" -s draft_shield_enabled="False" -s magic_spiralize="False" -s raft_fan_speed="0" -s magic_mesh_surface_mode="normal" -s raft_speed="30.0" -s infill_mesh="False" -s print_sequence="all_at_once" -s meshfix_keep_open_polygons="False" -s ooze_shield_dist="2" -s ooze_shield_angle="60" -s ooze_shield_enabled="False" -s multiple_mesh_overlap="0.15" -s prime_tower_wipe_enabled="False" -s prime_tower_flow="100" -s prime_tower_position_y="200" -s wireframe_nozzle_clearance="1" -s prime_tower_position_x="200" -s prime_tower_size="0" -s support_extruder_nr_layer_0="0" -s adhesion_extruder_nr="0" -s raft_base_fan_speed="0" -s bottom_thickness="0" -s machine_max_jerk_e="5.0" -s skin_overlap_mm="0" -s raft_interface_jerk="20" -s wireframe_up_half_speed="0.3" -s raft_surface_acceleration="3000" -s support_interface_extruder_nr="0" -s support_line_distance="2.25" -s raft_interface_speed="22.5" -s material_print_temperature="230" -s raft_base_line_spacing="1.6" -s raft_surface_line_spacing="0.45" -s raft_surface_line_width="0.45" -s cool_fan_speed="50" -s coasting_min_volume="0.8" -s prime_tower_enable="False" -s raft_surface_thickness="0.2" -s raft_interface_line_width="0.8" -s machine_disallowed_areas="[]" -s infill_sparse_density="20" -s jerk_support_interface="20" -s support_bottom_distance="0" -s raft_surface_layers="2" -s raft_airgap="0.57" -s raft_margin="15" -s brim_line_count="18" -s extruder_prime_pos_z="0" -s acceleration_prime_tower="3000" -s skirt_gap="3" -s extruder_prime_pos_y="0" -s conical_overhang_angle="50" -s retraction_retract_speed="25.0" -s extruder_prime_pos_x="0" -s support_tower_diameter="3.0" -s retraction_enable="True" -s support_interface_line_distance="0.45" -s raft_acceleration="3000" -s support_bottom_height="1" -s support_interface_enable="False" -s support_join_distance="2.0" -s material_standby_temperature="200" -s support_bottom_stair_step_height="0.3" -s support_xy_distance_overhang="0.2" -s support_xy_distance="0.7" -s wall_line_width_x="0.45" -s raft_base_thickness="0.24" -s support_top_distance="0.19" -s machine_nozzle_tip_outer_diameter="1" -s support_pattern="concentric" -s support_offset="0.2" -s jerk_travel_layer_0="30.0" -s support_type="buildplate" -s cool_lift_head="False" -s cool_fan_speed_min="0" -s retraction_prime_speed="25.0" -s cool_min_speed="8" -s cool_fan_enabled="False" -s travel_avoid_distance="0.625" -s jerk_print_layer_0="20" -s jerk_support_infill="20" -s raft_interface_fan_speed="0" -s jerk_topbottom="20" -s jerk_wall_x="20" -s acceleration_enabled="False" -s jerk_wall_0="20" -s magic_fuzzy_skin_enabled="False" -s jerk_enabled="False" -s infill_mesh_order="0" -s raft_base_speed="22.5" -s speed_travel="120" -s layer_0_z_overlap="0.22" -s machine_max_acceleration_y="9000" -s acceleration_skirt_brim="3000" -s acceleration_travel_layer_0="5000.0" -s travel_avoid_other_parts="True" -s acceleration_print_layer_0="3000" -s jerk_infill="20" -s acceleration_support_interface="3000" -s raft_surface_jerk="20" -s acceleration_support_infill="3000" -s acceleration_wall_0="3000" -s machine_head_with_fans_polygon="[[0, 0], [0, 0], [0, 0], [0, 0]]" -s retraction_combing="all" -s acceleration_infill="3000" -s wireframe_fall_down="0.5" -s speed_equalize_flow_max="150" -s speed_equalize_flow_enabled="True" -s top_bottom_pattern="concentric" -s skirt_brim_line_width="0.45" -s cool_fan_full_layer="0" -s cool_min_layer_time_fan_speed_max="10" -s speed_slowdown_layers="2" -s skirt_brim_speed="20" -s support_enable="False" -s conical_overhang_enabled="False" -s speed_travel_layer_0="40.0" -s raft_jerk="20" -s speed_print_layer_0="20" -s raft_interface_thickness="0.30000000000000004" -s infill="0" -s magic_fuzzy_skin_point_dist="0.8" -s coasting_speed="90" -s support_infill_extruder_nr="0" -s speed_prime_tower="60" -s speed_support_infill="60" -s raft_base_jerk="20" -s speed_wall_x="40" -s machine_max_jerk_xy="20.0" -s extruder_prime_pos_abs="False" -s gradual_infill_step_height="5.0" -s speed_wall_0="20" -s switch_extruder_retraction_speed="20" -s retraction_hop="1" -s retraction_hop_only_when_collides="False" -s retraction_hop_enabled="False" -s acceleration_wall_x="3000" -s coasting_volume="0.064" -s machine_min_cool_heat_time_window="50.0" -s retraction_amount="8" -s skirt_line_count="1" -s retraction_count_max="90" -s prime_tower_line_width="0.45" -s machine_max_feedrate_e="25" -s material_flow="105" -s machine_height="200" -s raft_base_acceleration="3000" -s switch_extruder_retraction_amount="16" -s material_bed_temperature="105.0" -s jerk_skirt_brim="20" -s machine_heat_zone_length="16" -s infill_wipe_dist="0.1125" -s travel_compensate_overlapping_walls_x_enabled="True" -s infill_overlap_mm="0.0675" -s machine_max_acceleration_e="10000" -s infill_pattern="cubic" -s alternate_extra_perimeter="False" -s material_print_temp_wait="True" -s retraction_hop_after_extruder_switch="True" -s speed_layer_0="20" -s raft_surface_speed="30.0" -s skirt_brim_minimal_length="250" -s layer_height="0.2" -s infill_line_distance="6.75" -s machine_name="FDM Printer Base Description" -s z_seam_type="shortest" -s retraction_extrusion_window="8" -s xy_offset="0" -s infill_sparse_thickness="0.2" -s speed_topbottom="30.0" -s retraction_extra_prime_amount="0" -s support_area_smoothing="0.6" -s travel_compensate_overlapping_walls_0_enabled="True" -s raft_base_line_width="0.8" -s wireframe_straight_before_down="20" -s support_line_w [... cut ...]
it under the terms of the GNU Affero General Public License as published by
the Free Software Foundation, either version 3 of the License, or
(at your option) any later version.

This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU Affero General Public License for more details.

You should have received a copy of the GNU Affero General Public License
along with this program.  If not, see <http://www.gnu.org/licenses/>.
`

fieldOfView, 2016-09-04:
Well, there is certainly something strange about the model. For one the normals are not all facing the same direction; some point out, some point in. Here's the model in MeshMixer. The red-ish faxes are facing the wrong way.
[image]

Ofcourse it would be great if Cura handled this better, and most often it does.

fieldOfView, 2016-09-04:
Make the "Union Overlapping Volumes" setting visible, and set it to False (instead of the default True). That fixes it for me.

I'm gonna leave this issue open so our slicer guru can see if the current behavior is correct (and my fix is the correct fix).

pulcher, 2016-09-04:
Thanks, so it probably my lack of understanding.  Pretty new at all in this arena.  I have no doubt you guys get a lot of this.  Would there be a better place for me to report this sort of thing?

Ghostkeeper, 2016-09-05:
Once in a while we get a false positive of someone thinking there is a bug in Cura but it was actually a faulty model. We don't really mind (at least I don't), since it is usually easy to find the problem.

It's not the normals at fault. CuraEngine doesn't listen to them. The octagon in the linked zip-file has some extra walls in the middle, but the one in the picture and video doesn't (X-ray view shows up blue and white, as normal).

I just tried it in Cura 2.1 (the only version I happen to have installed on this computer) and it sliced fine. So I'm thinking something recently broke.

BagelOrb, 2016-09-05:
Some of the meshfixes settings are optimized to fix most faulty models, but can introduce problems at places where there is no problem. Try fiddling around with the settings in the mesh fixes category.
@fieldOfView already showed that that can solve your problem.

closed 2016-09-05 by BagelOrb