Ultimaker/Cura #973: cura 2.3 Retraction speed can not change

QoooQ, 2016-09-05:
cura 2.3 Retraction speed can not change
max is 25 mm/s why  bug? 
i need up speed to  Retraction
i use custom FDM printer

Drayson80, 2016-09-05:
Short term solution:
Please refer to https://ultimaker.com/en/community/22396-cura-23-beta-awesomeness-and-some-issues?page=3#reply-155404

But of course, this has to be fixed professionally :-)

Ghostkeeper, 2016-09-06:
Fixed in 6909f88ce6d3ea335fc2bd2f159fe64c2105aa21.

I've now made it so that there is no limit by default (other than light speed), and it is up to the printer definitions to set a limit if they have one. Note that most printers will have a limit in software, at least those that are based on Marlin.
Ultimaker's printers all have a limit of 45 in their (default) firmware.

closed 2016-09-06 by Ghostkeeper

Drayson80, 2016-09-09:
I´m not sure that this fixed the topic - gave the yesterdays new open beta a try but the warning at retraction is still too low (25) so that e.g. 40mm/s is red...

Ghostkeeper, 2016-09-09:
What I did was remove the limit for every printer, except for Ultimaker machines of which we know that the limit is 45 in their firmware (so 25 was a wrong limit).

I had also verified with Daid that the intended warning value is 25, for the simple reason that most feeders are unable to reach higher speeds than that.
40mm/s shouldn't be red unless the printer you're using has specifically lowered the maximum, in which case it is intended. It should be yellow since it is higher than 25.