Ultimaker/Cura #974: 2.3 Beta. Slicing Issues -  Slow or Not Slicing

innovo3d, 2016-09-05:
[image]

On my custom printer, Innovo INVENTOR, it hangs on the slicing and doesn't finish.

I loaded the Ultimaker 2+ printer and it finished the slicing but it was very slow.  In 2.1.2 it sliced automatically right away, but in Beta 2.3 it either doesn't work or is very slow.

thopiekar, 2016-09-05:
Your Cura's logs would be useful. It is possible that it throws some error messages, which could explain this behavior.

innovo3d, 2016-09-05:
Here you go!
http://pastebin.com/cejtppme

nallath, 2016-09-06:
Could you give us the .log files? (https://github.com/ultimaker/cura#logging-issues)

nallath, 2017-07-27:
Closing this because of lack of activity. A lot of things have changed since then, so this is probably no longer an issue.

closed 2017-07-27 by nallath