Ultimaker/Cura #975: Issues on Cura 2.1.3

printam3d, 2016-09-07:
Hello,
There some features and issues that former version of Cura 14.06 had or did very well:
- Split object into parts is a good function in Cura 14.06 that Cura 2.1.3 doesn't have
- reset all object positions just refreshes the position of the objects but doesn't reposition them so as to fit on the buildplate like it was done in Cura 14.06
- after importing an object in Cura if I want to rescale an object by percentage of the original I can do this in the beginning, but if I need to rotate it and then rescale you will notice that the percentages are not 100% on the 3 axis but whole other values.
[image]

I'm using Cura 2.1.3 with Ultimaker 2+

awhiemstra, 2016-09-07:
> Split object into parts is a good function in Cura 14.06 that Cura 2.1.3 doesn't have

Split object into parts was unfortunately dropped because the previous implementation could not really be re-used. It will hopefully come back some time in the future.

> reset all object positions just refreshes the position of the objects but doesn't reposition them so as to fit on the buildplate like it was done in Cura 14.06

We still lack a good built platform packing solution. This will also be fixed in the future.

> after importing an object in Cura if I want to rescale an object by percentage of the original I can do this in the beginning, but if I need to rotate it and then rescale you will notice that the percentages are not 100% on the 3 axis but whole other values. 

This has been fixed in 2.3.

By the way, next time please create a single issue per missing feature/bug. That makes it simpler for us to repsond to and avoids discussions getting off-topic. Thanks!

ieol, 2016-09-10:
Hi,
trying to use Cura 2.1.3 because it have some nice futures... but it continue to hang  on "Preparing to slice..." for files that on Cura 15.04.6 have no problem to load and slice...

Ultimaker 2
Windows 7 Pro 64bit
DELL T5600 ram 16Gb Quadro4000  Xeon E5-2650

Where is Version 2.3?? on Ultimaker site the last version is 2.1.3

fieldOfView, 2016-09-10:
Look below the big download button.

Make sure you (backup and) remove the cura folder in c:\users\[you]\AppData\Local\cura. There's a bug in the profile upgrade system.

ieol, 2016-09-11:
Hi,
it is a Beta version... no way ... I don't install beta version... :)
But thank to share the info about the bug on profile upgrade system.... I never take care to delete the Cura folder before... but now.. for sure I will delete it on any future installation...

IMHO It would be appropriate for those who are responsible for updating the software to consolidate the same rather than continue to add options

Ghostkeeper, 2016-09-11:
Note that the bug in the profile upgrade system only exists when upgrading from 2.1.3 to the 2.3 beta currently online (which has since been fixed also). It doesn't concern upgrading from version 15.04.6 to 2.1.3.

closed 2017-11-14 by ChrisTerBeke

fieldOfView, 2018-10-06:
"Split into parts" is part of the "Mesh Tools" plugin which is available in the Toolbox in Cura 3.5