Ultimaker/Cura #979: Print monitor/Print Controller UI discussion

kakaroto, 2016-09-08:
This is to have a discussion about the print monitor UI and/or whether we want to add control to it, or if we want a separate Print Controller tab instead.
This is to continue the discussion that was started in issue #70, more specifically, started at  https://github.com/Ultimaker/Cura/issues/70#issuecomment-245358634

To summarize: do we want to add controls to the print monitor. If Yes/No, what additional information or controls do we want to add to it. If no, do we want to add a Print Controller tab, similar to the print monitor, but only for manual control of the printer. 
Also, what UI design should be used, anyone has any suggestions/mockups/etc.. Let's discuss it before we implement it.

nallath, 2016-09-08:
I've already ensured that the architecture supports it. If we use PrinterOutPutDevice any printer that works with monitor tab (right now USB and the octoprint stuff made by @fieldOfView) 

I would like to have a way to add controls. I don't think that adding a third tab (next to monitor & settings) is a good idea. We could add a way to switch between monitoring & controling of sorts?

nickthetait, 2016-09-09:
How about hiding the controls once a print has started, then revealing them if it is paused or once the print completes?

fieldOfView, 2016-09-09:
I'll make some mockups after the weekend of what I have had in mind for a while.

tenten8401, 2016-09-10:
@nickthetait could always just grey them out instead of hiding completely.

kakaroto, 2016-09-12:
@tenten8401 I think the idea to hide them is to avoid the clutter while printing. Even if you're not printing, if you don't want to control the printer, then having the controls there will still clutter the window. If the user wants to take control, he could click the "control button" which would 'slide in' a controls drawer. That's the way I'd see it I guess.
I'm not sure it's clear so here's a little demo of how I imagined it : http://www.leotheme.com/demo/prestashop14x/?template=leo_gamegear
If you click on the little gear icons on the left hand side of the page, you see some controls slide out. 

That's kind of neat. I don't know if it would look good, or if it would be doable with qml, or if it's an absurd UI design idea, but we're here to discuss ideas, so... :)

kakaroto, 2016-09-14:
@fieldOfView did you make any progress on the mockups  ? 
I've been working a bit on this, and I've run into a couple of UI design concerns for the print monitor.

The USB printing plugin will give us a "Print with USB" entry. I guess clicking on it would do the whole scan+connect+print, I thought of maybe changing it into "Connect to USB" and once it's connected, the menu item becomes "Print with USB".. but I'm not sure, it sounds bad from a user-experience perspective. It can stay as is and just to the connect+print instead, but if you have other ideas, I'm open to suggestions.

The other problem is that, as soon as we add a USB output device, the print monitor will always say "Connected with USB" and will show the print monitor icon with the green check mark on it, even if it's not yet connected. My issue is that it's misleading to the user, so how do we want to proceed ? Create a new icon (or simply keep the 'no printers' icon for the monitor) and show "USB devices available" or "Connect to USB" instead of the "Connected via USB" ? Should I add a 'Connect' button next to the Pause/AbortPrint buttons ? Any better ideas ?
I also don't like how you can't print form the print monitor, you would have to switch to the settings tab then click on the print arrow to see choices, then select "print with USB" from the list, then click on the button itself. How about making the "Abort Print" button into a "Print" button, so you can launch the print directly from the print monitor panel?

Another issue I'd like to discuss is the case of having more than one connected printer. Right now, if you have 2 printers, you'll see 2 "Print with USB" icons in the menu, that's confusing for the user. Shouldn't it be `"Print with USB: %s"  % device` instead ? It's not a big deal right now though because I will first have a single print item (for the autodetect), regardless of how many printers are connected, but eventually, if we recognize printers via their serial number (see discussion in #70), then I'd like to have multiple menu items, one per 'known printer' and one for the autodetect. Would you agree? And in that case, how to differentiate? like I said `"Print with USB: %s" % device` (where 'device == 'Autodetect' in the case of autodetect), or do you think there's a better way to show the information to the user ?
The other issue is of course with the print monitor which currently uses `Cura.MachineManager.printerOutputDevices[0]` automatically. Imagine you want to use octoprint for a remote printer, but you have a printer connected in usb, you can print to octoprint, but the print monitor will show you the info of your locally usb-connected printer instead.
I think if more than one printer is available, the user should be able to choose which one to monitor (usb + octoprint, switch from one to the other to monitor them all at the same time), so, since the print monitor already has a "Printer" combobox, why not add (or actually replace it with) a "Connection" combobox under it, just like we have "Material" and "Profile" comboboxes in the settings panel. And in that case, what should be its content ? the device.getName() ? if yes, we should add the device id to the name instead of having simply "USB Printing", and for the Octoprint plugin, it uses the 'key' variable, which I don't know what it is/represents, so I don't know if that would be an appropriate value to show in the combobox. Do you have other suggestions ? Actually, I'm not sure about the "Connection" word, maybe keep the combobox labeled as "Printer", but just use it for listing the printer output devices instead.

Any thoughts ?

Ghostkeeper, 2016-09-15:
> I also don't like how you can't print form the print monitor, you would have to switch to the settings tab then click on the print arrow to see choices, then select "print with USB" from the list, [...]

Completely agree with this. I've been annoyed by this before but never "consciously" enough to report it.

fieldOfView, 2016-09-16:
I'm sorry I'm taking my sweet time to answer and make some mockups about the direction Cura is going to. I only work for UM 3 days in the week officially but there's quite a lot on our plates here atm.

fieldOfView, 2016-10-18:
This is what I was thinking about to extend the sidebar print monitor:
[image]

Not all output devices will support all features in this proposal. The print-monitor would have to be smart enough to show only those parts that are applicable to the currently active output device. The "Printer Control" section would of course be disabled during a print.

Alternatively these additions to the print monitor could be implemented in a "machine action" in a separate window. I'm not a fan of that solution though, because those windows are modal and make the workflow with the print monitor less clear.

fieldOfView, 2016-10-18:
Also see this related issue on configuring/initiating

fieldOfView, 2017-02-13:
For the record, here's the Lulzbot take (proposal):
[image]

from: https://code.alephobjects.com/T304

mikeq3, 2017-09-08:
I like it! That was one thing I was really missing on Cura 2.7 was the ability to manually extrude, move the head around, manually preheat and see the temp on a graph. Some times you need to heat the nozzle to change or prime new filament.

mikeq3, 2017-09-08:
Can I beta test the Ultimaker one? I have the Ultimaker Original.

fieldOfView, 2017-09-09:
Just to be clear, there is no "ultimaker" implementation of this. Yet.

mikeq3, 2017-09-09:
The one you show above says ultimaker 2. I thought you had something you were testing.

Ghostkeeper, 2017-09-15:
@fieldOfView used to be in charge of UX for Cura. He would first make a semi-interactive design before it actually gets implemented. This never got implemented. We have a ticket for it, pointing to this comment on Github where it got requested again: https://github.com/Ultimaker/Cura/issues/2278#issuecomment-325633492

mikeq3, 2017-10-21:
Any chance on getting fieldofview's mockup implemented. The lack of manual control is a real show stopper for me adopting Cura3.0.3/2.7.  I actually slice in Cura 3.0.3 and save the gcode. Then I load it in Cura 15.04 to print to my Ultimaker Original over USB. Manual control of pre-heating and extruding is part of my workflow. I'm not familiar with QT and qml or I would have coded something myself.

nallath, 2017-10-23:
It's not something that Ultimaker is going to invest time in. So it's only going to be in there if someone else makes it (and creates a nice pull request for it)

closed 2017-11-13 by ChrisTerBeke

Ghostkeeper, 2017-11-20:
Most of this has been fixed via https://github.com/Ultimaker/Cura/pull/2669, so we consider it fixed now. If there's anything else that can be added, let's do that in separate requests to prevent this issue from becoming a boiling pot for all USB printing related issues.