Ultimaker/Cura #980: JSON code conventions

nickthetait, 2016-09-09:
I'm starting work on getting JSON files standardized. I've based the formatting on https://github.com/Ultimaker/Meta/blob/master/code_conventions.md Please take a peek at current state here: https://code.alephobjects.com/diffusion/CT/browse/master/resources/definitions/fdmprinter.def.json
Then click 'view raw file' to see formatting even clearer.

This is using some custom settings within Eclipse/JSON Editor Plugin then hitting the Edit -> Format JSON button. Want to check that this looks sensible before going further. Additionally should I update this for all the existing printer definitions?

fieldOfView, 2016-09-10:
Having the code on your own system makes it a bit harder for use to compare the changes. The automated change seems to have made some strange decisions eg in line 849, 990, 1149, ...

We know we have quite a few violations against our own code conventions, but have decided against doing sweeping codestyle fixes because it makes tracking changes harder. In this case I don't think the benefits outweight the disadvantage of breaking git blame.

NB: we don't have official code conventions for JSON or qml. Especially the latter leads to discussion from time to time (semi colons, yes or no?)

nallath, 2016-09-12:
I still think that we should have some official code conventions for both of them. On the forum someone also suggested using jsonmin to be able to add pseudo comments to the json file (which we then strip before parsing it as json)

nickthetait, 2016-09-13:
So should I make a PR with these changes for just fdmprinter.def on master branch? Are they likely to be accepted?

nickthetait, 2016-09-13:
Here's a better view of just the diff https://code.alephobjects.com/rCTdc77fbb671e4fd9f204a9793f56634a38a82f509#be0cfced

thopiekar, 2016-09-13:
@nickthetait Alternatively just create a PR and set it here, if already done. We can discuss in the PR whether it is worth to merge it. (And often it is :wink: )

PR #993 (2016-10-14): Matchup JSON formatting for fdmprinter with code conventions
Fixes #980
closed 2016-10-14 by Ghostkeeper