Ultimaker/Cura #981: Crash when window size was changed and you open the menu

sedrubal, 2016-09-11:
When the window size of cura was changed and you quickly click on `File` menu entry (first entry in menu bar) cura often crashes. First it freezes for 1 second then the monitor turns black and then Windows shows a notification that the Intel HD graphics driver was stopped because of an error.

Windows 7 64Bit
Cura: 2.1.3 64Bit

Ghostkeeper, 2016-09-11:
Can't reproduce this using my NVIDIA GeForce GTX 850M. I suspect it is specific to the Intel drivers, or your Intel driver.

Did you try updating your graphics driver?

Perhaps you can give us the output of DxDiag. That may help a bit with this sort of problem. To do that, follow these steps:
* Press Windows+R keys.
* Type dxdiag and press enter.
* Wait until the progress bar on the bottom of the window has finished.
* Click on Save All Information...
* Save it somewhere and attach the file here.

But I hope updating the graphics driver already works. I tried looking around for known bugs with Qt and Intel drivers crashing but couldn't find any needles in that haystack.

awhiemstra, 2016-09-12:
The Windows Intel drivers are notoriously bad. This specific issue also sounds like a driver issue, since Windows reports the driver crashes. I am afraid there is little we can do to fix this, both because we have no way of fixing the driver and because the relevant code is in Qt and outside of our control, so there is no simple workaround we can make.

sedrubal, 2016-09-12:
Thanks for your help. I think you're right. It might be a driver issue.

I uninstalled the driver and reinstalled it and now I didn't experienced it again.

This is the content of DXDiag:

```
------------------
System Information
------------------
Time of this report: 9/12/2016, 22:53:13
       Machine name: WS19
   Operating System: Windows 7 Enterprise 64-bit (6.1, Build 7601) Service Pack 1 (7601.win7sp1_ldr.160516-0600)
           Language: German (Regional Setting: German)
System Manufacturer: MSI
       System Model: MS-7823
               BIOS: BIOS Date: 06/07/13 10:49:15 Ver: 04.06.05
          Processor: Intel(R) Core(TM) i5-4670K CPU @ 3.40GHz (4 CPUs), ~3.4GHz
             Memory: 16384MB RAM
Available OS Memory: 16070MB RAM
          Page File: 2778MB used, 28777MB available
        Windows Dir: C:\Windows
    DirectX Version: DirectX 11
DX Setup Parameters: Not found
   User DPI Setting: 96 DPI (100 percent)
 System DPI Setting: 96 DPI (100 percent)
    DWM DPI Scaling: Disabled
     DxDiag Version: 6.01.7601.17514 64bit Unicode

------------
DxDiag Notes
------------
      Display Tab 1: No problems found.
        Sound Tab 1: No problems found.
          Input Tab: No problems found.

--------------------
DirectX Debug Levels
--------------------
Direct3D:    0/4 (retail)
DirectDraw:  0/4 (retail)
DirectInput: 0/5 (retail)
DirectMusic: 0/5 (retail)
DirectPlay:  0/9 (retail)
DirectSound: 0/5 (retail)
DirectShow:  0/6 (retail)

---------------
Display Devices
---------------
          Card name: Intel(R) HD Graphics 4600
       Manufacturer: Intel Corporation
          Chip type: Intel(R) HD Graphics Family
           DAC type: Internal
         Device Key: Enum\PCI\VEN_8086&DEV_0412&SUBSYS_78231462&REV_06
     Display Memory: 1888 MB
   Dedicated Memory: 384 MB
      Shared Memory: 1504 MB
       Current Mode: 1920 x 1200 (32 bit) (59Hz)
       Monitor Name: PnP-Monitor (Standard)
      Monitor Model: unknown
         Monitor Id: 
        Native Mode: unknown
        Output Type: Unknown
        Driver Name: igdumdim64.dll,igd10iumd64.dll,igd10iumd64.dll,igdumdim32,igd10iumd32,igd10iumd32
Driver File Version: 10.18.0014.4264 (English)
     Driver Version: 10.18.14.4264
        DDI Version: 11
       Driver Model: WDDM 1.1
  Driver Attributes: Final Retail
[... 585 lines cut ...]
---------------
EVR Power Information
---------------
Current Setting: {5C67A112-A4C9-483F-B4A7-1D473BECAFDC} (Quality) 
  Quality Flags: 2576
    Enabled:
    Force throttling
    Allow half deinterlace
    Allow scaling
    Decode Power Usage: 100
  Balanced Flags: 1424
    Enabled:
    Force throttling
    Allow batching
    Force half deinterlace
    Force scaling
    Decode Power Usage: 50
  PowerFlags: 1424
    Enabled:
    Force throttling
    Allow batching
    Force half deinterlace
    Force scaling
    Decode Power Usage: 0
```

closed 2016-09-12 by sedrubal