Ultimaker/Cura #982: How to completely remove ALL on Ubuntu. (2.3.0 PPA)

opensourcefan, 2016-09-11:
I'm having some issues with 2.3 and would like to do a complete fresh install. However after sudo apt-get remove or purge many profiles and folders etc remain behind. When I reinstall after that it opens to previous machines and profiles that I've setup. I would like every instance of Cura gone so I can start fresh.  How do I remove this completely?

Ghostkeeper, 2016-09-11:
Cura's settings are stored in `~/.config/cura` and in `~/.local/share/cura`. Deleting those two folders should amount to a fresh install.

I don't know where the PPA installs its files on your system though. Perhaps a combination of uninstalling with apt and deleting those folders remove everything? I'd expect Ubuntu to clean up after itself nicely.

opensourcefan, 2016-09-11:
That's exactly what I've just done, we'll see if that works. I tried un-installing every which automatic way including Synaptic but none of those are cleaning it up. You're right, being Ubuntu this "should" be cleaning up properly without manually searching for files left behind. Usually [purge] would remove config files as well but it didn't in this case.

opensourcefan, 2016-09-12:
Manually removing those files helped. It installed as new which was nice. All the issues though are still present.

closed 2016-09-12 by Ghostkeeper