Ultimaker/Cura #984: Crash when adding new printer. (Ubuntu 16.04) (2.3.0 PPA)

opensourcefan, 2016-09-11:
I'm having all kinds of issues with 2.3.0 after updating it today. It's not functional at all with many different interactions causing freezing. I also get tons of warnings and debugs on the start of Cura from Terminal. 

This is the Terminal output when I try and add a new printer. It results in a hang with the gui becoming greyed out. Force Quit is required. 

`file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
qml: 
qml: 
qml: 
2016-09-11 15:50:17,858 - WARNING - .../UM/Settings/ContainerRegistry.py (addContainer [219]): Container of type <class 'UM.Settings.InstanceContainer.InstanceContainer'> and id Custom FDM printer_variant already added
2016-09-11 15:50:17,917 - DEBUG - /usr/lib/cura/plugins/CuraEngineBackend/CuraEngineBackend.py (_onBackendQuit [412]): Backend quit with return code 127. Resetting process and socket.
2016-09-11 15:50:17,918 - DEBUG - .../UM/Backend/Backend.py (_createSocket [206]): Previous socket existed. Closing that first.
`

opensourcefan, 2016-09-12:
Here's another output from Terminal re this. In this instance I tried to add the Prusa_i3 printer. Upon hitting the add printer button it froze. 

`file:///usr/share/cura/resources/qml/AddMachineDialog.qml:105: ReferenceError: sortedMachineDefinitionsModel is not defined
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
qml: 
2016-09-11 16:56:53,905 - WARNING - .../UM/Mesh/MeshBuilder.py (reserveFaceCount [158]): Had to convert 'num_faces' with int(): 2684.0 -> 2684 
qml: 
qml: 
2016-09-11 16:56:54,086 - DEBUG - .../UM/Mesh/MeshData.py (calculateNormalsFromVertices [339]): Calculating normals took 0.0005602836608886719 seconds
2016-09-11 16:56:54,087 - DEBUG - /usr/lib/uranium/plugins/FileHandlers/STLReader/STLReader.py (read [35]): Loaded a mesh with 8052 vertices
Terminated
`

fieldOfView, 2016-09-12:
The freezes sound like the QT deadlock. @thopiekar, what version of pyqt is used with the ppa?

thopiekar, 2016-09-12:
@fieldOfView: My stable PPA comes with no PyQt5 at all. Installing 2.3 from there uses the PyQt5 version, which comes with Ubuntu. In case Ubuntu Xenial is used:
* http://packages.ubuntu.com/de/xenial/python3-pyqt5
@opensourcefan Which Ubuntu version are you using?

opensourcefan, 2016-09-12:
I'm on 16.04 (Xenial)

thopiekar, 2016-09-12:
Are you familiar with advanced package managers, like Synaptic?

opensourcefan, 2016-09-12:
Yes, I have it installed. Don't use it much though.

opensourcefan, 2016-09-12:
Current Python version is 2.7.12 which I think was installed from the previous 2.1.3 cura PPA.

thopiekar, 2016-09-12:
Please enabled my master PPA and install PyQt 5.6 from there (not Cura, etc.).
If it works I can copy PyQt 5.6 from there into stable.

thopiekar, 2016-09-12:
In case it works, I should better remove all packages for Ubuntu versions older than Xenial (16.04) just before other people get problems with that.

opensourcefan, 2016-09-12:
Is this what you want enabled?
ppa:thopiekar/cura-master

I've enabled it but can find PyQt 5.6 anywhere. I have apt-get updated.

thopiekar, 2016-09-12:
There should be PyQt 5.6:
* https://launchpad.net/~thopiekar/+archive/ubuntu/cura-master/+packages?field.name_filter=&field.status_filter=published&field.series_filter=xenial

thopiekar, 2016-09-12:
I'm getting these locks (in settings UI) on my notebook, too. I'll try it also on my own..

thopiekar, 2016-09-12:
[image]

thopiekar, 2016-09-12:
Now no locks here anymore. @opensourcefan, can you confirm this?

opensourcefan, 2016-09-12:
I just installed pyqt5, how would you like me to check exactly. I'll start it up and see, if there's something specific you would like me to check please advise.

opensourcefan, 2016-09-12:
I've added and taken away a few printers and all is well, no more lock ups.

opensourcefan, 2016-09-12:
No more locks, seems good now.

thopiekar, 2016-09-12:
Ok, now just disable the master PPA again. I'll publish PyQt 5.6 in my stable PPA soon.

thopiekar, 2016-09-12:
@opensourcefan Thank you for your feedback :+1:

opensourcefan, 2016-09-12:
Thank you for the excellent work!

thopiekar, 2016-09-12:
No problem :ok_hand:

closed 2016-09-12 by Ghostkeeper

thopiekar, 2016-09-12:
Just kept this open so I don't forget to remove all the packages for Ubuntu < 16.04.

However, done that now, so let's wait for the blame that packages for Ubuntu Wily or Vivid are not there anymore :see_no_evil: