Ultimaker/Cura #985: Warnings and Debugs on startup. (Ubuntu 16.04) (2.3.0 PPA)

opensourcefan, 2016-09-12:
Here's my Terminal output on start up, seems like I'm getting a bunch of warnings and plenty of Debugs. My install is quite unstable and get many freezes doing different things. Please close this if it's not useful, just wanted to make sure nothing is being missed. 

[Cura_2.3.0_Start_Output_Ubuntu_16.04.txt](https://github.com/Ultimaker/Cura/files/466467/Cura_2.3.0_Start_Output_Ubuntu_16.04.txt)

fieldOfView, 2016-09-12:
The warnings and debug statements are all knows/expected. Please see @Ghostkeeper's comment here:
https://github.com/Ultimaker/Cura/issues/938#issuecomment-241296409

closed 2016-09-12 by fieldOfView

awhiemstra, 2016-09-12:
Since you are running on 16.04, I assume you are running @thopiekar's PPA or a self compiled version. This most likely means that the freezes you mentioned are a deadlock bug in PyQt that we found and that in the meantime got fixed. We build a patched version of PyQt ourselves that includes that fix. In addition PyQt 5.6 or higher fixes the issue. You might want to try and upgrade.

opensourcefan, 2016-09-12:
Yes, thank you. All is well now after we confirmed the pyqt5 fix on the other issue. Look forward to playing with 2.3.0.

thopiekar, 2016-09-12:
@awhiemstra, discussed about this in another PR. The problem was that PyQt 5.5.1 was used. I thought that the Ubuntu maintainers already fixed the deadlock, but they didn't.
I added 5.6 to my PPA now and will remove all packages for all Ubuntu versions older than Xenial (16.04) so reports on that issue won't appear that much in the future.
By the way Yakkety (16.10) comes with 5.7 and works well so far.