Ultimaker/Cura #987: Crashes on startup: OpenGL error

sedrubal, 2016-09-12:
When starting Cura on Windows Server 2008, Cura crashes (window turns white and the well known Windows message appears "Cura has stopped working") with this additional info:

```
Problemsignatur:
  Problemereignisname:	APPCRASH
  Anwendungsname:	Cura.exe
  Anwendungsversion:	0.0.0.0
  Anwendungszeitstempel:	54467a5f
  Fehlermodulname:	OPENGL32.dll
  Fehlermodulversion:	6.3.9600.17415
  Fehlermodulzeitstempel:	54505402
  Ausnahmecode:	c0000005
  Ausnahmeoffset:	00000000000100d2
  Betriebsystemversion:	6.3.9600.2.0.0.16.7
  Gebietsschema-ID:	1031
  Zusatzinformation 1:	f8d9
  Zusatzinformation 2:	f8d96e1dacaea22c5c067c750d41e574
  Zusatzinformation 3:	8e99
  Zusatzinformation 4:	8e997621dfe1bc65cda95a36917ccb3a
```

Is this a dup of #283 or #981?

gurkih, 2016-09-12:
@sedrubal mixed up the OS-version, were actually using Windows Server 2012R2

sedrubal, 2016-09-12:
I just found out that this only happens on RDP sessions. On the local session it is working without errors.

sedwards2009, 2016-09-13:
I quickly googled this up: https://www.opengl.org/discussion_boards/showthread.php/198132-OpenGL-and-Microsoft-Remote-
Desktop

It looks like you can't expect OpenGL to work very well over RDP, and if you try you might get a fall-back software GL driver. This is most likely the problem. Cura requires a real OpenGL driver.

You could try using VNC instead of RDP if you need to work remotely.

gurkih, 2016-09-13:
Unfortunately vnc is not an option as were using a terminalserver setup. Is there another thing we could try? Can i somehow tell Cura to use some kind of 'software mode' ?

thopiekar, 2016-09-13:
According to http://stackoverflow.com/questions/27704529/deploying-qt5-on-windows-without-hardware-acceleration , you need to set QT_OPENGL=software as environment variable.
They also mention a opengl32sw.dll.. Never tried that on Windows..

closed 2016-09-17 by thopiekar

RealDekkia, 2020-05-07:
I stumbled over this problem today.

You're constantly pumping out updates with features nobody asked for, but you can't fix this bug?

Ghostkeeper, 2020-05-11:
I don't think we ever built a feature nobody asked for.

I'm all for building features and fixing the bugs that the most people ask for ("democracy of the loudest voice" if you will). And considering that, I think the group of people slicing via RDP connection falls right off the radar, compared to... The group of people having issues with small gap filling, or rotation+non-uniform scaling, or any of the other 1200 open issues.