Ultimaker/Cura #988: Cura 2.3.0 hangs on 'preparing to slice'

dwm101, 2016-09-14:
I've just installed Cura 2.3 to use with my Ultimaker 2 and no matter what model I open the system just sits on 'Preparing to slice' but doesn't actually do anything. Any thoughts?

[image]

sorvis, 2016-09-15:
The update to 2.3 broke slicing for me too. I found this error message repeated in my engine log:

`/usr/bin/CuraEngine: error while loading shared libraries: libArcus.so.2: cannot open shared object file: No such file or directory`

Ghostkeeper, 2016-09-15:
Dwm101 seems to be running into a setting problem: The quality-changes profile is None. I have a sneaking suspicion it might be caused by a faulty upgrade of his settings from version 2.1. You might try deleting (or moving away) this folder: `C:\Users\<dwm101>\AppData\Local\cura`. You'll lose all your preferences and profiles doing so, though.

Sorvis is running into a build problem... libArcus isn't being included/installed properly?

dwm101, 2016-09-15:
Yeah, I was wondering that actually- when I open 2.3 it mentions it's imported my custom profiles which could well be screwing things up. Will have a clear out and try again.

Ghostkeeper, 2016-09-15:
Known bug then which has since been fixed.

closed 2016-09-15 by Ghostkeeper