Ultimaker/Cura #996: 2.3 stable from ppa on Ubuntu 16.04 won't slice

sorvis, 2016-09-15:
Within the past week or so the update from the PPA thopiekar/cura broke slicing. The other day the engine showed some error about a library but today after updating cura again its now showing at the top of the engine log:
```Calling engine with: ['/usr/bin/CuraEngine', 'connect', '127.0.0.1:49674', '-j', '/usr/share/cura/resources/definitions/fdmprinter.def.json', '']```

I took a look at the path /usr/bin/CuraEngine and I see that is invalid, but for me in ubuntu 16.04 it should be /user instead of /usr. How does that path get set? Could it be caused by a miss commit or is that caused by some mistake by the build the ppa draws on?

thopiekar, 2016-09-16:
Why is "/usr" called "/user" on your desktop? "/usr" is a default directory used on all Linux distributions to share content across all user accounts. Never heard about possibilities to change the name of that directory and if you really did this I expect that you also had to do some modifications to other system packages. So if you _really_ moved all content from "/usr" to "/user" then just move the content again or create a symbolic link.

thopiekar, 2016-09-16:
This worries me the most:
```
2016-09-16 09:57:04,729 - ERROR - .../UM/Logger.py (logException [53]): Exception: Job <CuraEngineBackend.StartSliceJob.StartSliceJob object at 0x7f3e60b2e9b0> caused an exception
2016-09-16 09:57:04,785 - ERROR - .../UM/Logger.py (logException [53]): Traceback (most recent call last):
2016-09-16 09:57:04,785 - ERROR - .../UM/Logger.py (logException [53]):   File "/usr/lib/python3/dist-packages/UM/JobQueue.py", line 122, in run
2016-09-16 09:57:04,786 - ERROR - .../UM/Logger.py (logException [53]):     job.run()
2016-09-16 09:57:04,786 - ERROR - .../UM/Logger.py (logException [53]):   File "/usr/lib/cura/plugins/CuraEngineBackend/StartSliceJob.py", line 147, in run
2016-09-16 09:57:04,786 - ERROR - .../UM/Logger.py (logException [53]):     self._buildGlobalSettingsMessage(stack)
2016-09-16 09:57:04,786 - ERROR - .../UM/Logger.py (logException [53]):   File "/usr/lib/cura/plugins/CuraEngineBackend/StartSliceJob.py", line 223, in _buildGlobalSettingsMessage
2016-09-16 09:57:04,787 - ERROR - .../UM/Logger.py (logException [53]):     if user_container.hasProperty(key,"value") or quality_changes_container.hasProperty(key,"value"):
2016-09-16 09:57:04,787 - ERROR - .../UM/Logger.py (logException [53]): AttributeError: 'NoneType' object has no attribute 'hasProperty'
```

thopiekar, 2016-09-16:
Doing:
```
thopiekar@hp:~$ rm -rfv .local/share/cura/ .cache/cura/ .config/cura/
'.local/share/cura/machine_instances/old/1/BQ+Hephestos+2.cfg' wurde entfernt                                                                                                                                                                                          
removed directory '.local/share/cura/machine_instances/old/1'                                                                                                                                                                                                          
removed directory '.local/share/cura/machine_instances/old'                                                                                                                                                                                                            
'.local/share/cura/machine_instances/BQ+Hephestos+2.stack.cfg' wurde entfernt                                                                                                                                                                                          
removed directory '.local/share/cura/machine_instances'                                                                                                                                                                                                                
removed directory '.local/share/cura/quality'                                                                                                                                                                                                                          
removed directory '.local/share/cura/profiles'                                                                                                                                                                                                                         
removed directory '.local/share/cura/machines'                                                                                                                                                                                                                         
'.local/share/cura/instance_profiles/old/1/BQ+Hephestos+2.curaprofile' wurde entfernt                                                                                                                                                                                  
removed directory '.local/share/cura/instance_profiles/old/1'                                                                                                                                                                                                          
removed directory '.local/share/cura/instance_profiles/old'                                                                                                                                                                                                            
removed directory '.local/share/cura/instance_profiles'                                                                                                                                                                                                                
removed directory '.local/share/cura/variants'                                                                                                                                                                                                                         
removed directory '.local/share/cura/extruders'                                                                                                                                                                                                                        
'.local/share/cura/cura.log' wurde entfernt                                                                                                                                                                                                                            
removed directory '.local/share/cura/user'                                                                                                                                                                                                                             
removed directory '.local/share/cura/materials'                                                                                                                                                                                                                        
removed directory '.local/share/cura/'                                                                                                                                                                                                                                 
'.cache/cura/definitions/mankati_fullscale_xt_plus' wurde entfernt                                                                                                                                                                                                     
'.cache/cura/definitions/innovo_inventor' wurde entfernt                                                                                                                                                                                                               
[... cut ...]
'.cache/cura/definitions/prusa_i3_xl' wurde entfernt
'.cache/cura/definitions/mendel90' wurde entfernt
'.cache/cura/definitions/fdmprinter' wurde entfernt
'.cache/cura/definitions/bq_hephestos_2' wurde entfernt
removed directory '.cache/cura/definitions'
removed directory '.cache/cura/'
'.config/cura/old/2/cura.cfg' wurde entfernt
removed directory '.config/cura/old/2'
removed directory '.config/cura/old'
'.config/cura/cura.cfg' wurde entfernt
removed directory '.config/cura/'
```
.. fixed the problem here.

sorvis, 2016-09-16:
I'm officially crazy you are right it is not /user on my desktop, but /usr on my desktop. Also my bash history doesn't show the commands I "thought" I ran last night to figure that out lol. Anyways deleting the config directories you did fixed the latest version of Cura and its slicing now.

Thank you for your help and thank for all the hard work you do at maintaining the ppa!

thopiekar, 2016-09-16:
No problem. A shame you also just removed the configs. Would be more useful if we had these broken configs. Hopefully someone else will get into this, too.

sorvis, 2016-09-16:
My laptop was broken the same way too. At lunch break I could compress those directories and upload them if no one else beats me too it.

thopiekar, 2016-09-16:
Great! Thanks!

Just open a terminal and type "cura". You will also see colored output, so critical errors are hovered in red: (But some errors like the one you see here are non-critical..)
[image]

thopiekar, 2016-09-16:
@sorvis Which printer do you own?

sorvis, 2016-09-16:
@thopiekar I have a Maker Select i3. I use Octoprint for printing though and only slice with Cura.

thopiekar, 2016-09-16:
(Got a OctoPrint plugin in my PPA)

sorvis, 2016-09-16:
[broken_cura_settings.tar.gz](https://github.com/Ultimaker/Cura/files/477293/broken_cura_settings.tar.gz)

Here is a tar of those three folders from my laptop where Cura slicing is also broken.

Which plugin is that? Does it integrate Cura with Octoprint?

fieldOfView, 2016-09-16:
https://github.com/fieldofview/OctoPrintPlugin
It lets you upload directly from Cura (without the need of an intermediate gcode file) and monitor the progress (including a webcam view)

thopiekar, 2016-09-16:
@sorvis Installable via: `sudo apt-get install cura-extra-plugin-octoprint`

sorvis, 2016-09-16:
That sounds really neat! Thanks the name of the package was giving me trouble finding it.

peteruithoven, 2016-09-18:
I had the same problem, removing the files in `.local/share/cura/ .cache/cura/ .config/cura/` also worked for me.

closed 2016-09-19 by nallath

plasmasolutions, 2016-09-23:
Same problem here, removing .local/share/cura/ did the trick. If someone would like to investigate the bug: here is the packed .local/share/cura/

[https://dl.dropboxusercontent.com/u/93280964/cura.tar.gz](https://dl.dropboxusercontent.com/u/93280964/cura.tar.gz)

Will keep it uploaded for 10 days...tried to upload it directly on github (with a zip archive) - no luck.

Greetings, 
Thomas