Ultimaker/Cura #997: Cura 2.1 Support problem

Jay-watson, 2016-09-16:
The support material starts off from the bed where it should start from slope of the model. This leads to wasted plastic and the print time increases. I like a lot of the new features in cura 2.1 but this is a show stopper.

[image]

thopiekar, 2016-09-16:
You can increase the angle to a higher value than the angle you use in your model.
The setting is available in the sidebar as "Overhang Angle".

Jay-watson, 2016-09-16:
Hi, i tried increasing but no difference. if I go beyond 90 deg it has no support at all. See below

[image]

thopiekar, 2016-09-16:
I see now more support? Probably I've got the wrong direction of the angle in mind..
You' already tried to decrease, isn't it?

Jay-watson, 2016-09-16:
i just tried decrease and the problem doesnt go away. Thanks for your interest here, I am going to have to go back to 15.04 it seems.

thopiekar, 2016-09-16:
Could you be so kind and upload the file?
Would like to reproduce it here 😉

nallath, 2016-09-16:
Have you tried the same with 2.3?

Jay-watson, 2016-09-16:
Here is the file that I have used to demonstrate the problem

[Support issue Cura.zip](https://github.com/Ultimaker/Cura/files/477397/Support.issue.Cura.zip)

Jay-watson, 2016-09-16:
Well done, problem doesn't exist wih Cura 2,.3 Beta

[image]

thopiekar, 2016-09-16:
👍

closed 2016-09-16 by thopiekar