Ultimaker/Cura #1004: Hangs at "preparing to slice" on Ubuntu 16.04

GaryWSmith, 2016-09-17:
Here is the output after starting Cura from the terminal and loading a model.

2016-09-17 14:14:18,185 - ERROR - .../UM/Logger.py (logException [53]): Exception: Job <CuraEngineBackend.StartSliceJob.StartSliceJob object at 0x7fc6c74e0470> caused an exception
2016-09-17 14:14:18,185 - ERROR - .../UM/Logger.py (logException [53]): Traceback (most recent call last):
2016-09-17 14:14:18,185 - ERROR - .../UM/Logger.py (logException [53]):   File "/usr/lib/python3/dist-packages/UM/JobQueue.py", line 122, in run
2016-09-17 14:14:18,185 - ERROR - .../UM/Logger.py (logException [53]):     job.run()
2016-09-17 14:14:18,186 - ERROR - .../UM/Logger.py (logException [53]):   File "/usr/lib/cura/plugins/CuraEngineBackend/StartSliceJob.py", line 147, in run
2016-09-17 14:14:18,186 - ERROR - .../UM/Logger.py (logException [53]):     self._buildGlobalSettingsMessage(stack)
2016-09-17 14:14:18,186 - ERROR - .../UM/Logger.py (logException [53]):   File "/usr/lib/cura/plugins/CuraEngineBackend/StartSliceJob.py", line 223, in _buildGlobalSettingsMessage
2016-09-17 14:14:18,186 - ERROR - .../UM/Logger.py (logException [53]):     if user_container.hasProperty(key,"value") or quality_changes_container.hasProperty(key,"value"):
2016-09-17 14:14:18,187 - ERROR - .../UM/Logger.py (logException [53]): AttributeError: 'NoneType' object has no attribute 'hasProperty'

thopiekar, 2016-09-17:
Looks like your configuration is broken, too. This is a duplicate to #996 .

closed 2016-09-17 by thopiekar

thopiekar, 2016-09-17:
@GaryWSmith, have you tried to remove your current configuration?
Also could you zip' it before removing it, so we could have more data to find the reason why that happens?

This would be great :+1:

GaryWSmith, 2016-09-18:
Hi Thomas,

Yes, removing the config file solved the problem.
Here is the zipped problematic ".config/cura" directory

Hope it helps

Gary

thopiekar, 2016-09-18:
Hmm, looks like sending the zip via email doesn't work :confused:

Glad to hear that Cura works again :+1: