Ultimaker/Cura #1005: How do I troubleshoot "ERROR - There are no file types available to write with!"

marhar, 2016-09-18:
Mac, Cura version 2.1.3

I have an STL file loaded, and I **think** good settings for a Monoprice Mini.

When I click 'Save to File', this error shows up in the log:

2016-09-17 22:36:17,901 - ERROR - There are no file types available to write with!

How can I troubleshoot?  I just need gcode.

thopiekar, 2016-09-18:
Hey @marhar ,
could you be so kind and retry with Cura 2.3? Thanks!
https://ultimaker.com/en/products/cura-software/list (Beta section below)

(Also helpful would be your OSX version.)

marhar, 2016-09-18:
Thanks, I'll check it out!
OSX Version is: 10.11.6 (15G31), El Capitan

marhar, 2016-09-18:
Great,  that fixed it, Now I can save to file and SD Card. Thanks!

closed 2016-09-18 by marhar

thopiekar, 2016-09-18:
Great! Happy printing 😉