Ultimaker/Cura #1006: Language package problem

twosilly-2B, 2016-09-19:
I try to change the language package, because my English is not very good, but I have found that the language package is not identified! "Fdmprinter.json.mo" is only the file is not identified!

[image]
[image]

I've also found that this is not recognized as a package of all languages.
[image]

Ghostkeeper, 2016-09-19:
Yes, we haven't yet translated the new texts for 2.3. They are currently being translated.

Some old texts are broken too, because we renamed the FDMPrinter file that had all the setting names and descriptions. That'll be repaired once we have new translations.

twosilly-2B, 2016-09-19:
OK! Thank you for your answers! 
@Ghostkeeper

closed 2016-09-19 by nallath