Ultimaker/Cura #1008: Support + Skirt/Brim = strange things

avogra, 2016-09-19:
Hi,
I stumbled over some strange behaviour, when Support and skirt/brim are used at the same time.
Version is Cura 2.3.0 win64 from 2016-09-08

Reference image: support enabled and skirt set to 10 lines, 3mm distance:
[image]

Looks fine fore me.
Now I reduce skirt distance to 0mm:

[image]

The skirt lines cross the support now and the support structures inside the part are vanished on the first layer.
Instead, I now use brim:

[image]

Looks the same as skirt with 0mm distance, no support inside the part.
This is Layer 2 also with brim:

[image]

Here is the support that should begin on layer 1.

Ghostkeeper, 2016-09-20:
Ah, known bug. We've fixed this in the meanwhile for the 2.3 release :smiley:

It's supposed to make a brim around the whole model, excluding support, for adhesion of the model to the bed if there is lots of support around it. We had changed that it would only make a brim around the model, then realised that the support now detaches from the bed, so now we make a brim around the model AND the support. The brim replaces the bit of support on layer 1 that it intersects with.

closed 2016-09-20 by Ghostkeeper

avogra, 2016-09-20:
If I reduce skirt distance to 0mm (my second picture) does it also touch the part then? In thas part, I would prefer a brim that only holds the support down, not touching my part. For now I solved that by setting skirt distance to 0,001mm. Is this still possible or maybe even with skirt distance 0mm?

Ghostkeeper, 2016-09-21:
Skirt distance to 0mm will make the skirt touch the model.

There is no way, as far as I know, to make a brim around the support but not around the model.