Ultimaker/Cura #1009: Support roof ignoring z-distance for thin structures

avogra, 2016-09-19:
Version is Cura 2.3.0 win64 from 2016-09-08

Look at the following screeshots. Layer height is 0.1mm, min. z-distance 0.1mm too.

This is the first layer of the part, that is to be supported:

[image]

This is 2 layers before the supported structure, which looks fine:

[image]

And this is the layer just below the part:

[image]

In the large area, there is no material as expected, but the thin structures have support layers here. The support roof will be bonded tightly. In the enlarged view of the same layer, there is a bit of support lines around the edge of the large area too. Is this intended?

[image]

Vandrasc, 2018-04-09:
Is this still an issue in the current version 3.2 or 3.3? If so, could you please attach a project file to the ticket so we can easily reproduce as well? Thanks

avogra, 2018-04-09:
No, the issue is gone in 3.2.1 :-)

closed 2018-04-12 by ianpaschal