Ultimaker/Cura #1010: Window positioning can destroy UI

kevans91, 2016-09-20:
Hello!

Running 2.3-BETA, installed yesterday from https://ultimaker.com/en/products/cura-software on a multi-monitor setup.

If I position the window across boundaries or outside of the boundaries of my outside monitors, the UI does odd things ranging from the menus not drawing except for the border or the whole window not drawing except for the border. This can be fixed by moving the window back in bounds, exiting, and re-opening Cura or manually fixing cura.cfg.

Simplest reproduction would be to modify cura.cfg in your profile, and in the [general] section, set window_left = -10 and then running Cura. The window should not draw properly.

kevans91, 2016-09-20:
Here's a set of logs to accompany this report: https://files.kyle-evans.net/cura-logs.zip

nickthetait, 2016-09-20:
Could you post a few screenshots too?

kevans91, 2016-09-20:
Hi,

Sure!

[image]
[image]

Further testing reveals that my initial diagnosis is wrong-ish. This problem occurs specifically with dual video cards, when part or all of the window is placed on a monitor rendered from what is presumably the "secondary" video card, and the actual coordinates of the window have no bearing on this -- it was mostly a coincidence that the monitor immediately to the left of my primary uses a different video card.

The first photo seems to be a result of the splash screen being drawn on my primary monitor (video card no. 1), while the *left** side of the window is on this monitor to the left of my primary (using video card no. 2)

* I've not tested to what extent this matters, but I do know that if I re-position my monitors such that the left side of the window is using video card no. 1, as the splash screen is, the window draws correctly.

thopiekar, 2016-09-20:
How about other 3D applications? Are they working well?
Multiple video cards means? AMD? Nvidia? Intel?
What happens if you just got one monitor plugged in?

Thanks!

kevans91, 2016-09-20:
Ah, my apologies for the lack of information:

Almost every application I use works well -- Blender, Cura 15.x included

For both tested machines, dual AMD Radeon HD 8570's

Just one monitor is fine, and making primary display one of the monitors
connected to the other video card reverses the problem. So far, it'll only
work well on monitors that are output of the same video card as the primary
display (whichever one the splash screen draws on, more generally). The
problem seems to come out purely when the window is crossing GPU boundaries.

thopiekar, 2016-09-20:
Ok, thank you for the info.

Could you provide us your log (it is in the AppData directory)? Hopefully it gives an error message where things to wrong.

Thanks again!

kevans91, 2016-09-20:
Hi,

You can find the full relevant logs here: https://files.kyle-evans.net/cura-logs.zip
Here's a gist of the stderr, which seems to be the only interesting file: https://gist.github.com/kevans91/504540782f0d54f64dd76580c9867f63

The menu drawing strangeness doesn't add anything to any of the three log files, but the application startup no draw scenario produces the stderr from there.

thopiekar, 2016-09-20:
Thanks! I'm sure someone can take a look on that tomorrow.

Ghostkeeper, 2016-09-21:
`_selection_pass` is `None` :confounded:

awhiemstra, 2016-09-21:
From the looks of it, all OpenGL rendering fails when dealing with the split-GPU boundary. This probably means it needs to be solved lower in the stack (Qt). To be honest, I have no clue how to fix it in the first place, I did not even know it was possible to render different screens with different GPUs.

This might be a related upstream bugreport: https://bugreports.qt.io/browse/QTBUG-50371?jql=text%20~%20%22multi%20gpu%22

kevans91, 2016-09-21:
@awhiemstra: Looking at the proposed patch, this seems like it would at least fix the more major of my two problems-- the whole window failing to draw. I don't really like their solution of "force windows to the main display," but I can live with it.

I'm attempting to track down another Qt5+opengl application to see if it has the same problem with menus failing to draw, because that doesn't seem to be addressed in the upstream patch. Qt4 did not seem to exhibit this behavior.

bond4u, 2017-02-05:
cura 2.3.1. on 2nd monitor, dropdown on 1st
[image]

kevans91, 2017-02-05:
Well, that certainly makes it a little more interesting. =)

We have a fairly nice stock of AMDs that I can test on to see if I can
narrow down to specific cards, but that would seem to imply multi-monitor
in general with select AMD machines.

I do know that with the video cards in my machine we never have problems
with in any other machines, and they get subjected to a lot of CAD and
other Engineering usage.

<https://cloud.githubusercontent.com/assets/66413/22625491/14789cd4-eba1-11e6-8d82-c11f1bd593af.png>

Ghostkeeper, 2017-02-06:
The drop-down on the other monitor is actually something that was fixed for 2.4. We didn't fix it, but the Qt developers did and we upgraded to a newer version :)

bond4u, 2017-02-07:
The funny  thing is that when I start cura, it appears on 2nd monitor and dropdowns appear on 1st.
But then I drag the window around so that part if it goes to 1st monitor and then drag it back to 2nd monitor and put it there then dropdowns appear on 2nd monitor.

bond4u, 2017-02-17:
Slightly different issue - submenu position seems to be restricted by first monitor.
[image]
My current setup. Cura is middle monitor.
[image]

Ghostkeeper, 2017-02-22:
What version are you using, Bond4u? We recently released 2.4.

bond4u, 2017-02-24:
Installed Cura 2.4.0. Looks good. Dropdowns and submenus are on same monitor as Cura.

Ghostkeeper, 2017-03-01:
Nice to have that confirmed.

The original issue still stands though (rendering broken across multiple monitors), so I'm keeping this issue open.

kevans91, 2017-03-01:
Thanks! I'll keep an eye as versions progress and let you know if upstream addresses it. =)

kevans91, 2017-05-25:
Closing this because I no longer have the capacity to test it -- work hardware got upgraded to GTX 1050s.

closed 2017-05-25 by kevans91