Ultimaker/Cura #1011: Cura 2.1.3 hangs at "Preparing to slice..." with Brim Platform Adhesion

adamsiem, 2016-09-21:
Cura 2.1.3 hangs at "Preparing to slice..." with Brim Platform Adhesion, but not with Raft or Skirt. Running on Mac OSX El Capitan 10.11.6.

BRIM FAILURE

[image]

BRIM FAILURE ENGINE LOG

```
Calling engine with: ['/Applications/Cura.app/Contents/MacOS/CuraEngine', 'connect', '127.0.0.1:49674', '-j', '/Applications/Cura.app/Contents/Resources/cura/resources/machines/prusa_i3.json', '-vv']

Cura_SteamEngine version DEV
Copyright (C) 2014 David Braam

This program is free software: you can redistribute it and/or modify
it under the terms of the GNU Affero General Public License as published by
the Free Software Foundation, either version 3 of the License, or
(at your option) any later version.

This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU Affero General Public License for more details.

You should have received a copy of the GNU Affero General Public License
along with this program.  If not, see <http://www.gnu.org/licenses/>.
Loading polygon setting machine_disallowed_areas not implemented...
Unrecognized data type in JSON: machine_platform_offset has type array
Loading polygon setting machine_head_polygon not implemented...
Loading polygon setting machine_head_with_fans_polygon not implemented...
Connecting to 127.0.0.1:49674
Connected to 127.0.0.1:49674
```

RAFT / SKIRT SUCCESS

[image]

adamsiem, 2016-09-21:
This issue is closed.

Reducing the Brim Width to 6mm, from the default 8mm, allowed the slicing to proceed and complete successfully.

**An error message describing this failure would be wonderful!**

closed 2016-09-21 by adamsiem

awhiemstra, 2016-09-21:
In 2.3 we display a proper message when this happens. So it was already fixed. :)