Ultimaker/Cura #1016: Solid wall thickness of a tube without using infill

vitalsparks, 2016-09-22:
This happened in earlier Cura also. To explain, I am using a simple example of a tube.

If the nozzle size is, say, 0.4mm and the wall thickness of the tube is 2.8mm, then this implies that Cura needs to lay down 7 concentric circles, assuming that the inner/outer wall thickness in Cura is set to 1.6mm or greater. But it does not - it sets both inner and outer thicknesses to 1.2mm (a total of 6 concentric circles), leaving a gap of 0.4mm between them. So it seems that it cannot work out that one of the walls could comprise 3 circles, and the other 4. 

In other words, to get a solid tube wall, this must be an even multiple of the nozzle diameter, not an odd one.

This matters to me because if I use 100% fill to close the gap, the printing time increases greatly, and the inner/outer surface quality is degraded, not to mention the significant noise and vibration during the infill which may also stress the machine somewhat due to the inertial forces required to do it.

You can see evidence of this problem by observing the layer view in Cura.

Ghostkeeper, 2016-09-23:
Duplicate of https://github.com/Ultimaker/CuraEngine/issues/393

It's a known issue. Workaround is to reduce the line width such that all gaps are more than one line width wide (so for a thickness of 2.8mm, a line width of, say, 0.39mm would be better).

closed 2016-09-23 by Ghostkeeper

CCS86, 2016-09-23:
Have you tested that theory @Ghostkeeper ?

I have tried cheating line width, and not gotten good results. I think Cura struggles with asymmetry in perimeter counts.

Ghostkeeper, 2016-09-23:
I tested the feature extensively. I must say though that I didn't test the case with a piece of like 0.41mm wide while the line width was 0.40mm on an actual printer (but the g-code is as expected in those cases too).

My closest test was a small spider web with strings of 0.5mm thick and a line width set to 0.35mm. The result was not ideal, since the surface tension of the material sometimes seemed to cause a dotted line instead of the 0.15mm wide line that was requested. In general, long lines of less than 0.3mm wide from a 0.4mm nozzle are rarely any good.

CCS86, 2016-09-24:
Yeah, for a bead that is free on one side at least, I try to stay .35mm or above on a .4mm nozzle. I think that in gaps which are bounded on both sides, laying in a narrower bead would be totally fine. You will be getting remelt of the border beads as well.