Ultimaker/Cura #1017: Failed to load STL file - uncaught exception

technoIazy, 2016-09-24:
Application Version: 2.3.0-beta (same file loads fine in 2.1.3 and 15.04.6)
Platform: OSX and Win7/64 (via Parallels)
Qt: ?
PyQt: ?
Display Driver: ?

Steps to Reproduce:
Loading of a specific STL file (see zip archive below

) fails in 2.3.0-beta, whereas it is loaded fine in 2.1.3 and 15.04.6. Tried to perform a clean reinstall of cura (deleting ~/Library/Application Support/cura); also tried to load the file using the win64 build (via parallels), same outcome.

Actual Results:
Message from the application: "Failed to load <path-to-file>.stl" (where <path-to-file> is the actual path of my file) at the bottom of the screen. The application remains perfectly usable after this message.
The following log is produced:
> 
2016-09-24 21:28:54,044 - DEBUG - UM/Mesh/MeshData.pyc (calculateNormalsFromVertices [339]): Calculating normals took 0.09467816352844238 seconds
2016-09-24 21:28:54,081 - DEBUG - .../uranium/plugins/FileHandlers/STLReader/STLReader.py (read [35]): Loaded a mesh with 1064844 vertices
2016-09-24 21:28:54,613 - CRITICAL - cura/CrashHandler.pyc (show [24]): An uncaught exception has occurred!
2016-09-24 21:28:54,613 - CRITICAL - cura/CrashHandler.pyc (show [24]): Traceback (most recent call last):
2016-09-24 21:28:54,614 - CRITICAL - cura/CrashHandler.pyc (show [24]):   File "cura/CuraApplication.pyc", line 462, in event
2016-09-24 21:28:54,614 - CRITICAL - cura/CrashHandler.pyc (show [24]):   File "UM/Qt/QtApplication.pyc", line 193, in event
2016-09-24 21:28:54,614 - CRITICAL - cura/CrashHandler.pyc (show [24]):   File "UM/Event.pyc", line 157, in call
2016-09-24 21:28:54,614 - CRITICAL - cura/CrashHandler.pyc (show [24]):   File "UM/Signal.pyc", line 146, in emit
2016-09-24 21:28:54,614 - CRITICAL - cura/CrashHandler.pyc (show [24]):   File "UM/Scene/SceneNode.pyc", line 120, in calculateBoundingBoxMesh
2016-09-24 21:28:54,615 - CRITICAL - cura/CrashHandler.pyc (show [24]):   File "UM/Scene/SceneNode.pyc", line 575, in getBoundingBox
2016-09-24 21:28:54,615 - CRITICAL - cura/CrashHandler.pyc (show [24]):   File "UM/Scene/SceneNode.pyc", line 642, in _calculateAABB
2016-09-24 21:28:54,615 - CRITICAL - cura/CrashHandler.pyc (show [24]):   File "UM/Mesh/MeshData.pyc", line 146, in getExtents
2016-09-24 21:28:54,615 - CRITICAL - cura/CrashHandler.pyc (show [24]):   File "UM/Mesh/MeshData.pyc", line 221, in getConvexHullVertices
2016-09-24 21:28:54,615 - CRITICAL - cura/CrashHandler.pyc (show [24]):   File "UM/Mesh/MeshData.pyc", line 213, in getConvexHull
2016-09-24 21:28:54,615 - CRITICAL - cura/CrashHandler.pyc (show [24]):   File "UM/Mesh/MeshData.pyc", line 205, in _computeConvexHull
2016-09-24 21:28:54,616 - CRITICAL - cura/CrashHandler.pyc (show [24]):   File "UM/Mesh/MeshData.pyc", line 320, in approximateConvexHull
2016-09-24 21:28:54,616 - CRITICAL - cura/CrashHandler.pyc (show [24]):   File "scipy/spatial/qhull.pyx", line 2235, in scipy.spatial.qhull.ConvexHull.__init__ (scipy/spatial/qhull.c:20312)
2016-09-24 21:28:54,616 - CRITICAL - cura/CrashHandler.pyc (show [24]):   File "scipy/spatial/qhull.pyx", line 331, in scipy.spatial.qhull._Qhull.__init__ (scipy/spatial/qhull.c:3668)
2016-09-24 21:28:54,616 - CRITICAL - cura/CrashHandler.pyc (show [24]): scipy.spatial.qhull.QhullError: Qhull error
2016-09-24 21:28:55,217 - ERROR - UM/Logger.pyc (logException [53]): Exception: Exception in mesh loader
2016-09-24 21:28:55,217 - ERROR - UM/Logger.pyc (logException [53]): Traceback (most recent call last):
2016-09-24 21:28:55,217 - ERROR - UM/Logger.pyc (logException [53]):   File "UM/Mesh/ReadMeshJob.pyc", line 55, in run
2016-09-24 21:28:55,218 - ERROR - UM/Logger.pyc (logException [53]):   File "UM/Mesh/MeshFileHandler.pyc", line 51, in readerRead
2016-09-24 21:28:55,218 - ERROR - UM/Logger.pyc (logException [53]):   File "UM/Mesh/MeshData.pyc", line 146, in getExtents
2016-09-24 21:28:55,218 - ERROR - UM/Logger.pyc (logException [53]):   File "UM/Mesh/MeshData.pyc", line 221, in getConvexHullVertices
2016-09-24 21:28:55,218 - ERROR - UM/Logger.pyc (logException [53]):   File "UM/Mesh/MeshData.pyc", line 213, in getConvexHull
2016-09-24 21:28:55,219 - ERROR - UM/Logger.pyc (logException [53]):   File "UM/Mesh/MeshData.pyc", line 205, in _computeConvexHull
2016-09-24 21:28:55,219 - ERROR - UM/Logger.pyc (logException [53]):   File "UM/Mesh/MeshData.pyc", line 320, in approximateConvexHull
2016-09-24 21:28:55,219 - ERROR - UM/Logger.pyc (logException [53]):   File "scipy/spatial/qhull.pyx", line 2235, in scipy.spatial.qhull.ConvexHull.__init__ (scipy/spatial/qhull.c:20312)
2016-09-24 21:28:55,219 - ERROR - UM/Logger.pyc (logException [53]):   File "scipy/spatial/qhull.pyx", line 331, in scipy.spatial.qhull._Qhull.__init__ (scipy/spatial/qhull.c:3668)
2016-09-24 21:28:55,219 - ERROR - UM/Logger.pyc (logException [53]): scipy.spatial.qhull.QhullError: Qhull error

Expected results:

The STL file should be opened since it does not seem to have formal errors in it.

Additional Information:
I also tried to convert the file to .obj using meshmixer, but this produced the same outcome.
The file is rather big and could have small geometries in it (sure way beyond the actual resolution of my UM2), which has been identified as a problem in previous versions. However, the very same file loads (and gets sliced) without any problems in 2.1.3 and 15.04.6. And because of my current workflow (scaling down a real word design to produce a scale model) I would need my favorite slicer to handle such files as it used to to in previous versions.

technoIazy, 2016-09-24:
I can't manage to upload a zipped archive with the stl, so I paste its dropbox link:
https://www.dropbox.com/s/as1qug9szhrsg0q/roof_stl.zip?dl=0

sedwards2009, 2016-09-25:
I'll look into it tomorrow. It will be something in the geometry which it is not liking and should handle better.

technoIazy, 2016-09-26:
Great, thanks. I did some other tests, I hope this can help with debugging. 
First, I have redesigned the part from scratch as a single part (previously it was an assembly of smaller parts and I was wondering if there were some tiny empty spaces within the structure). It does not open, either.
Then I've cut the part in several ways, to see if the problem comes from a particular geometry. 
I've put toghether a zip with the several tests: the files are prefixed with OK when I was able to load them in Cura 2.3-beta, and NOK when I had the "Failed to load" message:
https://www.dropbox.com/s/ere855bpl5hbp2m/geometry-tests-20160926.zip?dl=0
My understanding is that the problem is not necessarily related to a specific (tiny) geometry but rather to the combination of the geometry and overall size the size of the object. The same geometry works both in the X and Y direction as long as the size (depth) of the part/feature is not too big.

Ghostkeeper, 2016-09-26:
Could reproduce this.
* First I loaded the original file. That gave the error Itechnolazyl described.
* Then I loaded the file in a different editor (Cloud Compare). It showed up as a normal mesh, though it is very flat.
* I loaded the file in Cura 15.04, and it loads fine.
* I re-saved the mesh in Legacy Cura, in case there was anything wrong with the file format. The re-saved file still gave the same error.
* I scaled the Y-direction of the mesh x20 in Legacy Cura. This made the file load in Cura 2.3.
* I undid the scaling and instead rotated the file 45 degrees in Legacy Cura. This also made the file load fine in Cura 2.3.

My diagnosis: Our convex hull might have no surface. We apply some rounding down to 1mm, and in this case, the Y-direction is less than 1mm thick.

I saved my rotated file here if you'd like a workaround: http://dulek.net/work/tetto-piatto-fixed.stl
To get back to the original file, rotate the part -45 degrees using the blue handle in Cura 2.3 (though afterwards you probably want to lay it flat on the bottom if you want it to print properly).

ItechnolazyI, 2016-09-26:
The workaround works perfectly, thanks!
As for your diagnosis: could you confirm it with the other test/example parts I posted today? Some of those files work nice in cura 2.3 (those whose name begins with OK), yet they still have geometries less than 1 mm in the Y direction.

Ghostkeeper, 2016-09-26:
Indeed, all of the files with OK load fine, and the files with NOK don't load. I tried rotating the first one ("redesigned") around the X-axis for a bit, and then it loaded fine in Cura 2.3.

I've made an issue in our internal tracker. I don't know if it'll get enough priority for it to be included in version 2.3, that's up to our change control board.

ItechnolazyI, 2016-09-26:
Thanks. As a user I'm fine with the suggested workaround. However, as an additional info, I just checked and all the NOK files load fine in Cura 2.1.3 (osx). That made me think of a regression in 2.3 and that's why I opened the issue in the first place, after having read that 2.3 comes with an improved and much quicker way of loading the STL parts.

Ghostkeeper, 2017-05-01:
This was removed from our planning because it has been inactive for too long.

awhiemstra, 2017-05-02:
Actually, I think this is related to an issue @jackha has been working on, since it also seems to be a QHull error. This should be fixed for 2.6.

nallath, 2017-07-27:
As stated, should be fix, closing it.

closed 2017-07-27 by nallath