Ultimaker/Cura #1018: Firmware Update UM2 to UM2+ Bug in OS X Cura

cekuhnen, 2016-09-26:
OS X Al Capitan
Cura 2.1.3
MacPro 2008
Ultimaker 2 now a +
Connection via USB Cable front port MacPro

Cura OS X does not Update the firmware. It simply shows a progress bar and asks you to wait.

Cura Win 10 same machine does the firmware update in few seconds with a progress bar moving and a finale notification that the update was done!

thopiekar, 2016-09-26:
Please use 2.3.0 (currently beta). I remember that I've been working on that.
Now the firmware update tool should show you at least a useful message. Whereas Cura 2.1.x will hang, when a problem occurs in the background.

ChrisTerBeke, 2018-01-22:
It's been a long while since this without comments. Closing for now. Please re-open if it's still an issue in the latest stable Cura version.

closed 2018-01-22 by ChrisTerBeke