Ultimaker/Cura #1019: Support Prusa i3 Mk2

mrjohnc, 2016-09-26:
There are issues with Cura on the Prusa i3 MK2

http://shop.prusa3d.com/forum/original-prusa-i3-mk2-f23/cura-print-settings-t999.html

thopiekar, 2016-09-27:
Could you please make a summary? Thanks!

nallath, 2016-09-27:
If someone wants to update the definitions, patches are welcome. We (read; Ultimaker) won't dig through tons of forums to fix issues for printers that we don't develop. 

If someone wants me to personally put some time in this, send me a printer and i'll have a look at it.

mrjohnc, 2016-09-27:
Ok, I'm just a user, I don't understand what the issue is. I'll ask someone from Prusa Research to explain the issue. Can someone reopen this issue please?

nallath, 2016-09-27:
The point is, someone from prusa research should fix it if they want it to be in the cura distribution that we provide. We can't spend time on fixing things for products that we don't ship.

Ghostkeeper, 2016-09-28:
We can't really test it either, as we have no Prusa printers at this office.

Simsey, 2016-09-29:
Well, I'm using Cura 2.3 beta with settings from [this pull request](https://github.com/Ultimaker/Cura/pull/978) in combination with the quality profiles provided [here](https://github.com/Apsu/cura-i3-mk2) and get extremly accurate prints on the i3 MK2.

I guess the main issue might be a false "installation" of the printer definition or it is related to Windows, as I'm on macOS.

Apsu, 2016-10-01:
Hello folks,

I'm the author of the definition in that pull request and own the repo where it's being maintained. I'd be happy to answer any questions about the choices I made, but much of the discussion/reasoning is in the forum thread linked above.

If you look at my [repo](https://github.com/Apsu/cura-i3-mk2) there are installation instructions for Windows and Mac for the current release of 2.3, if that helps.

fieldOfView, 2016-10-02:
Thanks for chming in. I'm closing this issue, because we you now have a better place to discuss this: either the prusa forum, or the issue queue for @apsu's repo.

closed 2016-10-02 by fieldOfView