Ultimaker/Cura #1020: Changing temperature during print

MarPiRK, 2016-09-28:
It would be great if it was possible to change temperature during print, as it is possible in Repetier.

It'd be very useful for calibrating filament's optimal printing temperature, or to fine-tune it during print to obtain the best possible quality.

nallath, 2016-09-28:
Already implemented. Check the tweak at z script in the postprocessing plugin.

dintid, 2016-09-29:
As Nallath said.. it often doesn't work though for me.. never work if you make changes at layer 2.

Also find it in a stupid location: Extensions - Post-Processing - Modify G-Code

Ghostkeeper, 2016-09-29:
Can't we just not make the "Post Processing" menu into the actual button? Like, you just click Extensions -> Post-Processing.

dintid, 2016-09-30:
Ghostkeeper - you miss my main point: Post-processing is not associative to change settings at some layer.

Yes, it is for technical minded people, who's having a history in the 3D community, but a large (if not major) -part of Ultimaker consumerbase buys Ultimaker for ease of use / because it's not a requirment to be very technical minded.

I really don't consider it as post-processing either... for me it is just a setting, same as setting infill and I guess it's only really considered post-processing, as it is an extensions and the settings in it is applied on top of the "base" settings.. but that is extremely not relevant for the end-user.

To make it short: For most people the Change Setting at layer (notice I didn't call it Tweak) is just a normal setting and shouldn't be labeled with some unnecessary technical term like "post-processing". Certainly not be buried in nr. 3-4 submenu under scripts.

nallath, 2016-10-03:
UI wise I don't think you can leave it with the rest of the settings. The post processing scripts can have multiple instances with different settings, which is not what the "normal" settings are designed for.
Another point is that 99% of the people will probably not use post processing anyway, so sacrificing GUI "real-estate" seems like a bad idea.

dintid, 2016-10-03:
@nallath you rather misunderstood some of what I said: I did not request additional GUI real estate. 
I rather wanted the drop-down menu removed. Then have _mm_ section greyed out, if you input _layer_ value, or the other way around.
That or keep Drop-down but still use the greyed out method, to make it clear which one is active (yes, in addition to the drop-down, which the user easily miss).

**I disagree regarding GUI real-estate**... both because it doesn't take up a lot, but also because any improvement to intuitive usage is worth it.

Right now it is very counter-intuitive as you can input a value in layer, but it never activates unless you **also** change drop-down. No setting should ever require 2 inputs/actions for 1 variable.

**Normal setting**
It doens't matter how, where and why Tweak at Z is a post-processing. All GUI settings must always act the same.
This is rule number 1 of any GUI development.
As it is now the user gets comfortable with how basic settings works.. then go into Tweak at Z and now it works differently... that's just stupid.

**Post-processing:**
You so much miss my point here... 

This option should not be buried in there.

In fact, any references to post-processing should be removed from the program. 
This is not something the user should face.

Post-processing is a developer term and is not relevant.. for users it is just a setting to change heat after layer 1 and similar.

**Not advanced thing to change a setting**
This entire Change at layer is not an advanced thing.. it is a basic settings which most people I know use. 

Especially with all the new printers having special plates which requires some heat at first layer but then none at all after that.

BagelOrb, 2017-07-23:
I agree that burrying the post-processing settings in a totally different place than where normal settings are accessed is a bit weird.
Plugins should contribute settings to the normal place where settings are being displayed.

Ghostkeeper, 2017-08-06:
That's not going to work. Let's say you have 4 post-processing scripts active, in this order:

* One Search-and-Replace that changes all occurrences of `M104 S205` with `;PLACEHOLDER`.
* One Tweak at Z that sets the Printing Temperature to 210 at layer 10.
* One Tweak at Z that sets the Printing Temperature to 200 at layer 20.
* One Search-and-Replace that changes all occurrences of `;PLACEHOLDER` with `M104 S205`.

Now you have some problems when you want to put these settings in the main setting list. For one, each of these settings would have to be listed twice. No actually, infinitely many times if we want to retain the power of the current system. And they would have to be associated with each other: The "layer number" setting of Tweak at Z has to be associated with the correct printing temperature. The search-field has to be associated with the correct replace-field. Furthermore, there would have to be some ordering to these settings, since the "placeholder" trick to keep the ordinary `M104 S205` out of the firing range of whatever Tweak at Z is doing will not work if the ordering is incorrect. In fact, it could damage the g-code.

closed 2017-11-14 by ChrisTerBeke