Ultimaker/Cura #1021: Duplicated items appears on top of source part, or outside of print area

dintid, 2016-09-29:
When duplicating a single object it is placed next to the original, but when duplicating an object on the plate multiple times the duplicated part is often placed on top of an existing part where it "blinks" in and out. Causing much lag on the program.
Sometimes the part appears outside of printplate, but that is not often.

To clarify: it doesn't have to be the same piece I duplicate, but if you duplicate enough objects (3 or more) they start appearing on top of items allready there.

The above can be bandaided by moving the part manually, but it is a huge bother.

Ghostkeeper, 2016-09-29:
What version are you using?

We've fixed the blinking problem in 2.3 (or so we think).

The part can still end up outside of the build volume. We've planned a fix for that, but it's not going to be any time soon.

dintid, 2016-09-29:
I'm using 2.3

Ghostkeeper, 2016-09-29:
Could reproduce. I loaded a 4cm cube and duplicated it at random a couple of times.

dintid, 2016-09-30:
It also seems strange to me how it Duplicates to outside the printarea.
[image]

**Also make X copies is missing. Any plans to have this implemented again?**
I often need to make like 20 or more copies of some small piece, and it's really annoying to hit Duplicate this many time.. especially when many of them ends up on top of each other, so I have to manually sit and rearrange them.

nallath, 2016-10-03:
It is planned to be implemented again. I can't give any indications as to when though. 
Same goes for finding a location of an object inside buildplate when duplicating.

Ghostkeeper, 2016-10-03:
Also, we've had a feature request to re-introduce the "Clone X times" feature from legacy Cura. That should make it a bit easier to print 20 copies of something.

dintid, 2016-10-03:
Awesome for both things :)

nallath, 2017-07-27:
This has been fixed.

closed 2017-07-27 by nallath