Ultimaker/Cura #1023: Tweak at Z-layer 2 not working

dintid, 2016-09-29:
When using "post-processing" to tweak at layer 2 it doesn't work.

It often doesn't work when issuing changes later on as well, meaning it is rather unrealiable.

Ghostkeeper, 2016-09-29:
Hmm, getting segfault as soon as I enable the Tweak at Z script!

Ghostkeeper, 2016-09-29:
Can't reproduce. Changed temperature at layer 2 to 222 degrees C. It produces the following snippet:

    ;LAYER:2
    G0 F7200.0 X130.625 Y130.625 Z0.470
    M605 S0;stores parameters before tweaking
    ;TweakAtZ V5.1: executed at Layer 2
    M117 Printing... tw@L   2
    M104 S222.000000 T0
    ;TYPE:WALL-INNER
    G1 F2400.0 X92.375 Y130.625 E996.95628

fieldOfView, 2016-09-29:
@dintid, what version of Cura are you using?
The postprocessing plugin has received quite the overhaul in 2.3, which is currently in beta.

dintid, 2016-09-30:
It worked at first, but now it just stopped working for some reason? Anywhere I can look for error messages?

Ghostkeeper, 2016-09-30:
Yes, though the position of the log depends on your operating system. See for instructions here: https://github.com/Ultimaker/Cura#logging-issues

dintid, 2016-09-30:
I reproduced my error, and it is partly an error-40 (40cm from screen) but not entirely.

In the TweakZ we have to actively choose from a drop-down menu wheter we want to use Height in mm or Layer No.
As opposed to rest of program where latest inserted input is the active input.
This is the partly error40... I did not know this, as this behaviour is not the same as the rest of the program and is therefore not logical/expected.
Remember I asked to have the non-active option greyed out? This is precisely why :)
[image]

**Wrong layer change at**
When putting in the change at layer 2
[image]

The change is not at layer 2, but at layer 3.
[image]

Wheter this is because we start at Layer 0 or the programmer just want the change to take place at the **end of the input layer** I don't know, but the  result is the same: Not making sense for enduser/consumer.

The change at layer, really need to be at the start of the input layer, and not after the input layer.

Dim3nsioneer, 2016-09-30:
There was a conditional display for postproc script inputs in an earlier version (2.0 iirc) but it must have been dropped somewhere on the road...

fieldOfView, 2016-09-30:
@dim3nsioneer is right about the conditional display. It should still be there. I'll see if that needs fixing

dintid, 2016-10-01:
Might be because I'm Danish, so English isn't my primary language, but the comments on a conditional display is blowing past me... What do you mean? :)

fieldOfView, 2016-10-01:
It is supposed to work like this:
[image]
It is supposed to shown only Tweak Height _or_ Tweak Layer, not both.
(note that in this version I have expanded the abbreviations)

dintid, 2016-10-01:
I gathered that.. read my pointers on why that is bad? Mainly because it's opposed to every other function in Cura where the last entered input is the active input.. this is the only place where you actively have to choose the active setting.

Anyway, my question was what you meant by the "Conditional Display"? You meant the "Drop-down menu"?

The Change at layer occuring at the next layer must be a bug though.

Ghostkeeper, 2016-10-03:
With "conditional display" he meant that some settings only display under certain conditions. Tweak Height should only display if Trigger is set to "Height". Tweak Layer should only display if Trigger is set to "Layer No.". In your screenshot it displays both. Something is wrong there.

You don't have to choose one or the other (if you don't, it chooses a height of 5.0mm), and the last entered input is the active input. At least, it should be. If it isn't for Dintid, that's a bug.

dintid, 2016-10-03:
Thanks @Ghostkeeper .. the conditional behavior is what I really asked for here.
Both Height and Layer are always visible regardless of drop-down status.

I do have to choose, that is exactly what the problem is also about.

Problem for me was how I input layer-setting, while drop-down was listed at Height... so the change did not occur at the layer I wanted but at the height -> meaning at 5mm instead at 0.2mm.

Ghostkeeper, 2016-10-04:
So you'd like to have it a bit like Fan Full at Height/Layer, like the Fan Full at Layer setting overwrites the Fan Full at Height setting, but both are visible?

Our thought process was that the "Height" trigger is more common and less technical, and should therefore be the default, but a user can change it to a specific layer as well.

dintid, 2016-10-04:
I'm not sure entirely sure what you said/meant exactly @Ghostkeeper but I really just wanted it to be like @fieldOfView showed.. (except I'd like layer to be default if I could choose, but that is a minor point).

I originally asked as **mine shows both height and layer at once** regardless of my choice in drop-down menu like this, which I'm now told is a bug:
[image]

I rather liked the way it is shown in the clip @fieldOfView posted.. mine doesn't do that though.

**Height vs Layer**
I think that is very subjective how you like it.. I never use height as I see it as inaccurate.
I would think Layer was the most commonly used thing, but I might be wrong on this obviously.
I personally always use layer.. I use **view mode -> layer** and then pin-point the layer I want extra fan at, less heat and such... which leads to the next issue:

**Layer change action:**
A problem regarding accuracy here is when I find I want my heatbed to drop from 70 to 60 and hotend from 205 to 200 after **first layer**, meaning I configure it to **layer 2** the action is actually triggered at **layershift 2 -> 3**, meaning the effective _change is taking place at **layer 3 and not 2** as I specified_.

When using PEI surface where it likes to be at 130c for bed at first layer for ABS and then dropped to 90c.. it really means the world if the change is 1 layer later, which can easily be 30 minutes of much extra heat (elephants foot).

Ghostkeeper, 2016-10-05:
Right, I'd expect it to change the temperature _before_ layer 2 if you set it to tweak at layer "2", too. We'll have to take a look at that.

Ghostkeeper, 2016-10-05:
I tried to reproduce it, and there's definitely something weird going on with the layer it tweaks at. I've tried to set the temperature to 69 (teehee) at layer 2, to see what layer it would tweak at. Turns out it tweaked at layer 21 (right before layer 21).

It's not using the height, or not properly, either, since the height was still at 5.0 (though it was properly hidden on my screen) and my layer height was 0.1, so it's ending up somewhere around 2mm from the build plate instead of 5.

Changing the height and leaving it on "Height" results in g-code that shows it's being tweaked in the header but doesn't actually have a tweak anywhere. Changing the height to 1mm and then setting it back to tweak at layer makes it tweak at layer 21 again. Making the model larger still keeps the tweak at 21.

My diagnosis: I suspect that the engine is not sending the g-code layer-by-layer any more again. This was the problem previous time something like this happened.

dintid, 2016-10-05:
Ok, I'm not into the actual mechanics of gcode, meaning how it's supposed to be executed and such.
I have never used Height, but I can see that when I choose "Layer" from drop-down menu, and set layer at "layer 2" it changes at the start of "layer 3" (when viewing the gcode in notepad)... I'll wager it's because layer 1 is actually called layer 0.
Good old coding paradigm :)

The change does execute though.

Dim3nsioneer, 2016-10-05:
@Ghostkeeper If it is not too urgent you may assign it to me.

Ghostkeeper, 2016-10-06:
I've made an issue for it in our internal bug tracker (CURA-2560 for reference, nice round number!). There is quite a large chance that it'll get assigned for a later version. If that is the case, I'll assign it to you, @Dim3nsioneer. If it's going to be assigned for fixing in version 2.3, it'll need to get fixed in a few days anyway so you shouldn't bother. We must await the decision of the change control board today.

Ghostkeeper, 2016-10-06:
It was just allotted to version 2.4. I'll assign it to @Dim3nsioneer and give him a chance to fix it. If he doesn't get around to it, we'll fix it up ourselves in a few weeks/months for version 2.4.

Dim3nsioneer, 2016-10-10:
I got myself a bit a strange behavior yesterday... will have a look into it.

closed 2017-11-14 by ChrisTerBeke