Ultimaker/Cura #1024: Change Request: Tweak at Z should be a dedicated menu item!

dintid, 2016-09-29:
Not sure how I should post this, so please be patient with me.

For any non-programmer it is extremely illogical to have the **Tweak at Z layer** be placed as it is.
Any normal user/consumer is never going to go into: **Extensions -> Post-Procssing -> Edit Gcodes** and then find Tweak at Z.

**Why?**
1) Because normal users don't care if it is an extensions. Especially not since it is implemented from the start.
2) Normal user doesn't know what Post-Processing even means.
3) Normal users do not try post-processing anything. Ever.
4) For normal consumers, the **Edit Gcodes** implies modifying some codes, and will be avoided.
5) Lastly the user then has to **Add a script** in order to modify a layer. This action makes perfect sense for the programmer, but it just nonsense for normal users/consumers.

**My Suggestion:**
1) Make it a seperate menu point. A Main menu between View and Settings since its' function is to actually modify the current job.
2) If the first suggestion is too much, then at least make it a seperate point under **Extensions** instead of putting it under **Post-Processing** which is non-sensical for normal consumers.
3) Call the menu "Change settings at layer" or "Tweak at layer".

**TweakAtZ submenus are badly named**
Again, they make sense for a programming perspective and they I guess the names are actually the same variables used behind the scene (I havn't looked it up).. but/and they are not easy to sort through, even for experienced users.

Suggested Changes to names:
1) Do not use abbreviations, so no **Tw.Print...** 
2) **Don't call it Tweak** anything... call it what it is: a change. Instead it should be named: **Change Print Speed**
3)Visually seperate the options into sections using a line between each section and a headline, like: Tweak at - Print Speed - Flow Rate - Bed Adjustment - Hotend temperature - Fan speed.
4) Not a name, but when using the "Tweak Layer" option the "Tweak Height" option should be grayed over to assure the user that it is inactive.

fieldOfView, 2016-09-29:
I certainly agree on the ugly abbreviations.
For the rest of your requests: they are fair, but we won't have time to do this for a little while. Give it a couple of weeks to a month.

dintid, 2016-09-30:
I know they are not critical changes in a Beta, so not in a rush :)
I do however see the some/all changes as very important to have implemented before the Beta ends

fieldOfView, 2016-10-02:
We are currently in string freeze. That means all the texts in the application have been handed over to a translation agency, and we cannot introduce any new texts until after the 2.3 final release. That - and a lack of time - means we are stuck with the current implementation for the 2.3 release.

ChrisTerBeke, 2017-11-23:
This should be added to the post processing plugin right?

Ghostkeeper, 2017-11-23:
Ah yeah, one could report this to github.com/nallath/PostProcessingPlugin as well, but just like for Uranium and CuraEngine we tend to accept these sort of enhancements in the Cura issue page as well.

fieldOfView, 2017-11-23:
No, the main point of the issue is that the functionality of the Tweak At Z post processing script should be integrated into the main UI instead of being hidden away into Extensions -> Post Processing -> Modify GCode. Not a trivial task.

Ghostkeeper, 2017-11-23:
The plug-ins add their menu items themselves (there's just some convenience functions to add it to the Extensions menu), so it could add itself to a separate menu header if it liked. It's not trivial though, since we'd have to decide what to do when there are multiple of these; you still need that list of the scripts. We could just add menu items for "Add this script to the list".

ChrisTerBeke, 2017-11-23:
I don't think making this a main feature in the UI is something we want, as it's only for very specific use cases. The post processing plugin could indeed expose some menu items via the extensions menu to make is more accessible. This should always come from the plugin though because adding some UI elements depending on if a certain plugin name is loaded or not is super ugly.

closed 2018-11-26 by Appesteijn