Ultimaker/Cura #1025: Cura engine generate additional supports that rest in mid air when a second part is added

AlpNov, 2016-10-03:
With only the one part, cura engine can generate sensible support:

[image]

File is Deep_Space_Nine_ORing-FLO-printversion+support-3x.stl (attached in [model and config.tar.gz](https://github.com/Ultimaker/Cura/files/505395/model.and.config.tar.gz)). Scaled to 60%.

For this part, because the top is quite small and the cooling features like lift head to cool causes some oozing, I added a "wipe tower.stl" (attached in [model and config.tar.gz](https://github.com/Ultimaker/Cura/files/505395/model.and.config.tar.gz))  part that I scale in the z axis to create some additional cooling time for the top layer.

However when this part is added, cura also added some extra support to the first part:

[image]

In this layer for example, the two support on the side are not needed and the plastic extruded during this operation obviously doesn't stay in place. If they get dragged to other support area, they stick there, curl up and the nozzle could potential knock into them, knocking the other support pieces out of alignment.

[image]

I have tried changing a couple of support settings, but to no avail. Unfortunately I forgot to document these changes.

Attached Files:
[model and config.tar.gz](https://github.com/Ultimaker/Cura/files/505395/model.and.config.tar.gz)

AlpNov, 2016-10-03:
Update, curiously enough, scaling the wipe tower to just below the height of the main part worked to eliminate the extra support.

Ghostkeeper, 2016-10-03:
Note that rather than adding your own wipe tower, you can also adjust the cooling settings to make sure it gives enough time for a layer to cool: I suggest increasing the Minimum Layer Time setting a bit, maybe 15s or so. Also, set Minimum Speed to be equal to the wall speed and activate "Lift Head" for better cooling. If a layer takes less than 15s, it will then reduce the printing speed until it reaches Minimum Speed. If printing at Minimum Speed is still less than 15s, it will move the print head away and wait for a while. This way you won't have to print an additional tower, and it will reduce your printing time.

The bits of support are there because the spine has actual overhang there. Setting Support Overhang Angle to 80 seems to give good g-code if you don't want that support there. Playing with the X/Y and Z distances sometimes helps too, but in this case I couldn't get that right either.

There is no way to apply a morphological opening operation to support. We should introduce that some day.

AlpNov, 2017-01-09:
As scaling the wipe tower to equal or less than the height of the main part works to eliminate the extra support, but still serves the purpose of the wipe tower (adding extra cool time causes oozing in the nozzle so isn't a viable option). So I'm closing this issue, sorry for the late reply.

closed 2017-01-09 by AlpNov