Ultimaker/Cura #1026: Language choice is not what I want, I want to make love

jockeysoft, 2016-10-03:
Language choice is not what I want, I want to make love

jockeysoft, 2016-10-03:
I do not have the language I want to request to translate the file

Ghostkeeper, 2016-10-04:
What language would you like to have or make love in?

jockeysoft, 2016-10-04:
I would like to use the Chinese language.

fszhiyi@163.com
 
From: Ghostkeeper
Date: 2016-10-04 15:33
To: Ultimaker/Cura
CC: jockeysoft; Author
Subject: Re: [Ultimaker/Cura] Language choice is not what I want, and want to translate it (#1026)
What language would you like to have or make love in?

Ghostkeeper, 2016-10-04:
I've created an item in our internal tracker.

I can't give guarantees that it would get accepted!

jockeysoft, 2016-10-04:
OK Think You

fszhiyi@163.com
 
From: Ghostkeeper
Date: 2016-10-04 17:12
To: Ultimaker/Cura
CC: jockeysoft; Author
Subject: Re: [Ultimaker/Cura] Translate Cura into Chinese (#1026)
I've created an item in our internal tracker.
I can't give guarantees that it would get accepted!

jockeysoft, 2017-01-21:
I can translate the language file? Translation and modify those files please let me know if there is a configuration file better

fszhiyi@163.com
 
From: Ghostkeeper
Date: 2016-10-04 17:12
To: Ultimaker/Cura
CC: jockeysoft; Author
Subject: Re: [Ultimaker/Cura] Translate Cura into Chinese (#1026)
I've created an item in our internal tracker.
I can't give guarantees that it would get accepted!

kitchung, 2017-01-21:
Please note there are two variants of written Chinese, including traditional and simplified so it needs to be taken into account. https://en.wikipedia.org/wiki/Written_Chinese#Simplified_and_traditional_Chinese

jockeysoft, 2017-01-22:
What translates the Cura folder what file?

fszhiyi@163.com
 
From: kitchung
Date: 2017-01-22 01:17
To: Ultimaker/Cura
CC: jockeysoft; Author
Subject: Re: [Ultimaker/Cura] Translate Cura into Chinese (#1026)
Please note there are two variants of written Chinese, including traditional and simplified so it needs to be taken into account. https://en.wikipedia.org/wiki/Written_Chinese#Simplified_and_traditional_Chinese

jockeysoft, 2017-01-22:
Language translation file "*.po "

fszhiyi@163.com
 
From: kitchung
Date: 2017-01-22 01:17
To: Ultimaker/Cura
CC: jockeysoft; Author
Subject: Re: [Ultimaker/Cura] Translate Cura into Chinese (#1026)
Please note there are two variants of written Chinese, including traditional and simplified so it needs to be taken into account. https://en.wikipedia.org/wiki/Written_Chinese#Simplified_and_traditional_Chinese

Ghostkeeper, 2017-01-22:
Take the `*.pot` files rather than the `*.po` files. Remember to translate all the `*.pot` files from Uranium as well as Cura (all files that you'll see in the eventual Cura installation).

Also note that translating to Chinese requires Cura to use a different font, since the current font only has Latin and Cyrillic characters. Getting Uranium to switch between fonts based on the language is not trivial.

kitchung, 2017-01-23:
Poke around here: https://github.com/Ultimaker/Cura/tree/master/resources/i18n

Ghostkeeper, 2017-01-23:
There, but Uranium as well: https://github.com/Ultimaker/Uranium/tree/master/resources/i18n

There's these four files in total:

* https://github.com/Ultimaker/Uranium/blob/master/resources/i18n/uranium.pot
* https://github.com/Ultimaker/Cura/blob/master/resources/i18n/cura.pot
* https://github.com/Ultimaker/Cura/blob/master/resources/i18n/fdmextruder.def.json.pot
* https://github.com/Ultimaker/Cura/blob/master/resources/i18n/fdmprinter.def.json.pot

jockeysoft, 2017-01-23:
Think you!

fszhiyi@163.com
 
From: Ghostkeeper
Date: 2017-01-23 16:10
To: Ultimaker/Cura
CC: jockeysoft; Author
Subject: Re: [Ultimaker/Cura] Translate Cura into Chinese (#1026)
There, but Uranium as well: https://github.com/Ultimaker/Uranium/tree/master/resources/i18n
There's these four files in total:
https://github.com/Ultimaker/Uranium/blob/master/resources/i18n/uranium.pot
https://github.com/Ultimaker/Cura/blob/master/resources/i18n/cura.pot
https://github.com/Ultimaker/Cura/blob/master/resources/i18n/fdmextruder.def.json.pot
https://github.com/Ultimaker/Cura/blob/master/resources/i18n/fdmprinter.def.json.pot

jockeysoft, 2017-01-23:
Think You!

fszhiyi@163.com
 
From: Ghostkeeper
Date: 2017-01-23 06:57
To: Ultimaker/Cura
CC: jockeysoft; Author
Subject: Re: [Ultimaker/Cura] Translate Cura into Chinese (#1026)
Take the *.pot files rather than the *.po files. Remember to translate all the *.pot files from Uranium as well as Cura (all files that you'll see in the eventual Cura installation).
Also note that translating to Chinese requires Cura to use a different font, since the current font only has Latin and Cyrillic characters. Getting Uranium to switch between fonts based on the language is not trivial.

awhiemstra, 2017-01-23:
> Also note that translating to Chinese requires Cura to use a different font, since the current font only has Latin and Cyrillic characters. Getting Uranium to switch between fonts based on the language is not trivial.

If we have a translation that needs it, we can probably switch to use Noto Sans instead of OpenSans. Noto Sans includes fonts for CJK languages and a lot of others. The other option would be to simply not load a custom font when dealing with CJK, since we can reasonably assume the system font supports CJK.

jockeysoft, 2017-01-23:
I Poedit, non so cosa buon software di editing. Il contatto iniziale. Tu dici non sembra cambiare il cambiamento di carattere Poedit

fszhiyi@163.com
 
From: awhiemstra
Date: 2017-01-23 20:55
To: Ultimaker/Cura
CC: jockeysoft; Author
Subject: Re: [Ultimaker/Cura] Translate Cura into Chinese (#1026)
Also note that translating to Chinese requires Cura to use a different font, since the current font only has Latin and Cyrillic characters. Getting Uranium to switch between fonts based on the language is not trivial.
If we have a translation that needs it, we can probably switch to use Noto Sans instead of OpenSans. Noto Sans includes fonts for CJK languages and a lot of others. The other option would be to simply not load a custom font when dealing with CJK, since we can reasonably assume the system font supports CJK.

Ghostkeeper, 2017-01-24:
I hope you're not machine-translating Cura when you create the Chinese translation. We could do that ourselves, too, but the quality of machine translations is still terrible.

Ghostkeeper, 2017-01-24:
I always found PoEdit quite good translation software. But if you want an alternative, Virtaal is also quite all right.

jockeysoft, 2017-01-24:
Because I am in China, the translation will be handled properly. Google translation is good, much better than before. Virtaal first contact, it seems quite interesting.

fszhiyi@163.com
 
From: Ghostkeeper
Date: 2017-01-24 18:14
To: Ultimaker/Cura
CC: jockeysoft; Author
Subject: Re: [Ultimaker/Cura] Translate Cura into Chinese (#1026)
I always found PoEdit quite good translation software. But if you want an alternative, Virtaal is also quite all right.

jockeysoft, 2017-01-30:
Why not see CoreXY models? Which should be selected?

fszhiyi@163.com

Ghostkeeper, 2017-01-31:
> Why not see CoreXY models? Which should be selected?

I don't understand. I don't think we have a CoreXY printer added yet in Cura. But CoreXY/H-bot/Cartesian/Delta/whatever is irrelevant to Cura since we just provide the g-code and it's up to the printer firmware to translate that into proper motor instructions for their mechanical arrangement.

jockeysoft, 2017-01-31:
So that's the case. Yes, a few days I can upload the Chinese language to you. Hope to see the installation of the Chinese interface.

fszhiyi@163.com

jockeysoft, 2017-02-01:
Why not see the "cellular grid"?

fszhiyi@163.com

Ghostkeeper, 2017-02-02:
You're not seeing the 1cm grid of the build plate? Did I understand that correctly?

You might be using a printer that has a visual printer model that goes too high, so that it overlaps with the build plate. You could move the mesh a bit further down in the printer definition in that case.

jockeysoft, 2017-02-02:
Not that. Is a honeycomb filler

fszhiyi@163.com
 
From: Ghostkeeper
Date: 2017-02-02 20:16
To: Ultimaker/Cura
CC: jockeysoft; Author
Subject: Re: [Ultimaker/Cura] Translate Cura into Chinese (#1026)
You're not seeing the 1cm grid of the build plate? Did I understand that correctly?
You might be using a printer that has a visual printer model that goes too high, so that it overlaps with the build plate. You could move the mesh a bit further down in the printer definition in that case.

Ghostkeeper, 2017-02-02:
Cura doesn't have a honeycomb infill pattern yet. Try Triangles infill pattern for similar functionality.

The Honeycomb infill issue is still open: https://github.com/Ultimaker/Cura/issues/1287

jockeysoft, 2017-02-02:
I do not know where to translate. see attached

fszhiyi@163.com

jockeysoft, 2017-02-02:
What does "Jerk" mean? Can be understood as "high-speed"?

fszhiyi@163.com

nallath, 2017-02-03:
No. Jerk is the second order acceleration. It sort of means "The acceleration of the acceleration" or "How fast does the acceleration increase".

fieldOfView, 2017-02-03:
Wouldn't "The acceleration of the acceleration" be the "snap" or "jounce"?
velocity -> acceleration -> jerk -> snap (or jounce) -> crackle -> pop

Note that in 3d printing, jerk has a somewhat different definition than in pure physics (hence the unit being mm/s instead of mm/s<up>3</up>)

jockeysoft, 2017-02-03:
The original is so strange not baffled. Thank you!

fszhiyi@163.com
 
From: Jaime van Kessel
Date: 2017-02-03 18:55
To: Ultimaker/Cura
CC: jockeysoft; Author
Subject: Re: [Ultimaker/Cura] Translate Cura into Chinese (#1026)
No. Jerk is the second order acceleration. It sort of means "The acceleration of the acceleration" or "How fast does the acceleration increase".

thopiekar, 2017-02-03:
@fieldOfView Had this confusion also when getting an prototype working here and creating definitions for it in Cura.
In my point of view Cura should be more scientific correct and at least put jerk into `"` or something like that.

jockeysoft, 2017-02-03:
Does "Travel" mean air travel?

fszhiyi@163.com
 
From: Thomas Karl Pietrowski
Date: 2017-02-03 22:09
To: Ultimaker/Cura
CC: jockeysoft; Author
Subject: Re: [Ultimaker/Cura] Translate Cura into Chinese (#1026)
@fieldOfView Had this confusion also when getting an prototype working here and creating definitions for it in Cura.
In my point of view Cura should be more scientific correct and at least put jerk into " or something like that.

Ghostkeeper, 2017-02-06:
> Note that in 3d printing, jerk has a somewhat different definition than in pure physics (hence the unit being mm/s instead of mm/s3)

Exactly, jerk in this context is not third-order derivative of distance but some 3D-printing specific term.

Ghostkeeper, 2017-02-06:
> Does "Travel" mean air travel?

It's not "air travel" as in with an aeroplane.

It's moving the nozzle without extruding material.

jockeysoft, 2017-02-19:
Will my printer be unable to connect. How does the Hex file get? Such as my DIY printer.

fszhiyi@163.com

Ghostkeeper, 2017-02-23:
Please create new issues for this sort of thing. This issue is about translating Cura into Chinese.

It's currently impossible to upload custom hex files to your printer via the GUI. This feature was not re-implemented when we rewrote the GUI for Cura 15.06 (15.04 still had the feature). There is a workaround: Replace the Ultimaker Original hex file with your own hex file and in Cura add an Ultimaker Original printer, then try to upload the firmware. It simply grabs the hex file (based on the currently active printer) and sends it over the line. Can't guarantee anything though.

fieldOfView, 2017-02-23:
> It's currently impossible to upload custom hex files to your printer via the GUI.

Uh, yes it is. Since 2.3

nallath, 2017-02-23:
Yep. That was CURA-955. That was about ~5 months or so ago.

Ghostkeeper, 2017-08-09:
The original feature request is not implemented yet.

ianpaschal, 2018-02-21:
But it is now!

closed 2018-02-21 by ianpaschal