Ultimaker/Cura #1028: Mac Crash on Startup (Error log included)

SydneyMakers, 2016-10-04:
Purely based on the size of the error dump, i've Pastebin'd it.
http://pastebin.com/Hxsem5NQ

Ghostkeeper, 2016-10-05:
Could you post us Cura's log instead?

It's located in `~/Library/Application Support/cura/cura.log`.
The only hint this gives is that it crashes somewhere in the zlib module of CPython.

SydneyMakers, 2016-10-05:
http://pastebin.com/VBP07Cke
Done.

Ghostkeeper, 2016-10-06:
Thanks. Looks like it's not crashing on start-up though. You've loaded a mesh with 10452 vertices. Perhaps it's crashing during the file loading...?

larp-welt, 2016-10-20:
I have the same problem on Ubuntu 16.10 - using ppa:thopiekar/cura for installation! It's just a fresh install of Ubuntu 16.10.

Inside the shell, I get an error message (sorry, german system):

Speicherzugriffsfehler (Speicherabzug geschrieben)

That's saying something in the style of: Memory Access Error (Memory Dump written). But I don't know where to find that memory dump.

This is my log file:
[cura.log.zip](https://github.com/Ultimaker/Cura/files/542745/cura.log.zip)

larp-welt, 2016-10-23:
After todays update, I get some more errors... 

[cura.txt](https://github.com/Ultimaker/Cura/files/546646/cura.txt)

I renamed the cura.log to cura.txt for update.

Ghostkeeper, 2016-10-24:
> 2016-10-23 23:27:28,339 - CRITICAL - .../cura/CrashHandler.py (show [24]): An uncaught exception has occurred!
> 2016-10-23 23:27:28,339 - CRITICAL - .../cura/CrashHandler.py (show [24]): TypeError: unable to convert a Python 'str' object to a C++ 'bool' instance

These were fixed today by this pull request: https://github.com/Ultimaker/Cura/pull/1071

larp-welt, 2016-10-26:
The conversion-error is no gone. But the chrash still stays... :-(

larp-welt, 2016-11-01:
I found something new! When the crash happens, Ubuntu logs following error in dmesg:

`[67289.586331] cura[27049]: segfault at 0 ip 00007f28633e0c7f sp 00007ffe50ebd5e0 error 4 in libQt5XcbQpa.so.5.6.1[7f286333f000+fb000]`

I tried different graphic drivers, no change!

Ghostkeeper, 2016-11-01:
Ah. That helps a lot. It seems to be the same issue as the more recently reported issue here: https://github.com/Ultimaker/Cura/issues/1094#issuecomment-257543551

There has been quite a fruitful discussion over in that thread, with lots of logs and deep analysis. So I'll close this one as a duplicate (even though you were earlier). We should continue discussion over at https://github.com/Ultimaker/Cura/issues/1094 . Thanks!

awhiemstra, 2016-11-02:
The discussion related to the Ubuntu crash happes in #1094 but this one was originally about OSX. We should keep this open for the OSX issue.

Ghostkeeper, 2016-11-03:
Ah, sorry, I didn't see that larp-welt had switched to Ubuntu for that dmesg log.

jromero, 2016-11-14:
Any update on this issue? Just upgraded from 2.3 to 2.3.1 and Cura crashes immediately.

I removed the directory `~/Library/Application\ Support/cura/` to make sure it wasn't lingering from the previous version. 

But even when I start the app and it crashes there is no log file:
```bash
ls -al ~/Library/Application\ Support/cura/
ls: /Users/javier.romero/Library/Application Support/cura/: No such file or directory
```

Ghostkeeper, 2016-11-14:
Still in progress. There was a computer at work that reproduced something like that too, but the issue is basically upstream so we have no solution.

jromero, 2016-11-25:
UPDATE: I upgraded to MacOS Sierra 10.12.1. Same exact install worked without a problem after the upgrade.

closed 2017-11-13 by ChrisTerBeke