Ultimaker/Cura #1029: custom material profile causes critical error in console (+no material cost shown in gui)

sjoerdtimmer, 2016-10-05:
I have copied the standard ABS and called the duplicate "ABS+". I changed the default temp, color, and filament diameter but when the new material is selected the gui wont show any time/cost estimates any more. This is the console error:
```
2016-10-05 22:42:26,974 - CRITICAL - .../site-packages/cura/CrashHandler.py (show [24]): An uncaught exception has occurred!
2016-10-05 22:42:26,975 - CRITICAL - .../site-packages/cura/CrashHandler.py (show [24]): Traceback (most recent call last):
2016-10-05 22:42:26,976 - CRITICAL - .../site-packages/cura/CrashHandler.py (show [24]):   File "/usr/lib/python3.5/site-packages/cura/CuraApplication.py", line 483, in event
2016-10-05 22:42:26,976 - CRITICAL - .../site-packages/cura/CrashHandler.py (show [24]):     return super().event(event)
2016-10-05 22:42:26,976 - CRITICAL - .../site-packages/cura/CrashHandler.py (show [24]):   File "/usr/lib/python3.5/site-packages/UM/Qt/QtApplication.py", line 192, in event
2016-10-05 22:42:26,976 - CRITICAL - .../site-packages/cura/CrashHandler.py (show [24]):     event._function_event.call()
2016-10-05 22:42:26,976 - CRITICAL - .../site-packages/cura/CrashHandler.py (show [24]):   File "/usr/lib/python3.5/site-packages/UM/Event.py", line 157, in call
2016-10-05 22:42:26,977 - CRITICAL - .../site-packages/cura/CrashHandler.py (show [24]):     self._function(*self._args, **self._kwargs)
2016-10-05 22:42:26,977 - CRITICAL - .../site-packages/cura/CrashHandler.py (show [24]):   File "/usr/lib/python3.5/site-packages/UM/Signal.py", line 150, in emit
2016-10-05 22:42:26,977 - CRITICAL - .../site-packages/cura/CrashHandler.py (show [24]):     func(dest, *args, **kwargs)
2016-10-05 22:42:26,977 - CRITICAL - .../site-packages/cura/CrashHandler.py (show [24]):   File "/usr/lib/python3.5/site-packages/UM/Backend/Backend.py", line 187, in _onMessageReceived
2016-10-05 22:42:26,977 - CRITICAL - .../site-packages/cura/CrashHandler.py (show [24]):     self._message_handlers[message.getTypeName()](message)
2016-10-05 22:42:26,977 - CRITICAL - .../site-packages/cura/CrashHandler.py (show [24]):   File "/usr/lib/cura/plugins/CuraEngineBackend/CuraEngineBackend.py", line 364, in _onPrintTimeMaterialEstimates
2016-10-05 22:42:26,977 - CRITICAL - .../site-packages/cura/CrashHandler.py (show [24]):     self.printDurationMessage.emit(message.time, material_amounts)
2016-10-05 22:42:26,978 - CRITICAL - .../site-packages/cura/CrashHandler.py (show [24]):   File "/usr/lib/python3.5/site-packages/UM/Signal.py", line 150, in emit
2016-10-05 22:42:26,978 - CRITICAL - .../site-packages/cura/CrashHandler.py (show [24]):     func(dest, *args, **kwargs)
2016-10-05 22:42:26,978 - CRITICAL - .../site-packages/cura/CrashHandler.py (show [24]):   File "/usr/lib/python3.5/site-packages/cura/PrintInformation.py", line 81, in _onPrintDurationMessage
2016-10-05 22:42:26,978 - CRITICAL - .../site-packages/cura/CrashHandler.py (show [24]):     self._current_print_time.setDuration(total_time)
2016-10-05 22:42:26,978 - CRITICAL - .../site-packages/cura/CrashHandler.py (show [24]):   File "/usr/lib/python3.5/site-packages/UM/Qt/Duration.py", line 75, in setDuration
2016-10-05 22:42:26,978 - CRITICAL - .../site-packages/cura/CrashHandler.py (show [24]):     duration = round(duration)
2016-10-05 22:42:26,978 - CRITICAL - .../site-packages/cura/CrashHandler.py (show [24]): ValueError: cannot convert float NaN to integer
```

fieldOfView, 2016-10-06:
What version of Cura? 2.1.x or 2.3?
Could you post a pastebin of the file you created?

Ghostkeeper, 2016-10-06:
It also expects a unique GUID. Just generate a new one and replace that hex string in the file.

sjoerdtimmer, 2016-10-13:
I changed the GUID in .local/share/cura/materials/generic_abs+%232.xml.fdm_material but the problem still occurs. Strangely, it does not always happen but I have not yet figured out what makes it appear/disappear...

It is cura 2.1 built on
Thu 29 Sep 2016 03:11:37 PM CEST
I don't have the gitref anymore

sjoerdtimmer, 2016-10-13:
I just built 2.3 and the issue seems to be gone (I also cleared .local/share/cura so that may actually be the change). I'm going to close this issue now.

closed 2016-10-13 by sjoerdtimmer