Ultimaker/Cura #1030: Create Profile From Current Settings

sleemanj, 2016-10-06:
Cura 2.3, accept my apologies if this is already fixed, I couldn't see anything like it in the tracker.

When you adjust some settings, at least of a custom profile on-the-fly, and then select "Create profile from current settings..." option from the Profile drop down, it appears that only the currently uncommitted (not yet saved to profile) changed settings of the current profile are added to the new profile, instead of all the settings from the current profile plus the uncommitted ones.

The result is that various settings in the profile get reset to defaults, and thus the profile you created is not the same as the settings you were looking at (further, those uncommitted settings are lost)

This image shows a custom profile, there are no uncommitted changes, notice the Wall Thickness is set to 1.2mm in this profile
[image]

I make a change to the layer height (we see it is uncommitted because of the circular arrow)
[image]

I select Create profile from current settings...
[image]

(off-camera I call the new profile Normal Quality #5 and click the button)

The profile which gets created is different to the settings it was created from, notice the Wall Thickness has reverted to a default 0.8mm, instead of the 1.2mm it should have been, the layer height change however has been recorded.

[image]

fieldOfView, 2016-10-06:
Thanks for the report. I will log an issue in our internal tracker, and I will try to get it fixed before the 2.3 release.

The issue seems to be that when creating a new profile, Cura disregards the currently selected custom profile ("quality_changes container" in our code) and instead looks at the protected profile ("quality container" in our code) it was based on.

nickthetait, 2016-10-06:
Really great and detailed bug report :+1:

fieldOfView, 2016-10-06:
And it's been fixed, reviewed and tested. How's that for a turnaround? The fix will be part of the Cura 2.3 final release. 

Again, thanks for reporting.

closed 2016-10-06 by fieldOfView